f0cal/google-coral #1: install.sh throws an error if system python is not 3.5

robmarkcole, 2019-03-15:
When following https://coral.withgoogle.com/tutorials/accelerator/ I received an error in line 116 of the `install.sh` script, as it requires python **3.5**. I just ran the command with system python (3.7) and did not receive the error, so you might want to amend the script.

jtgans, 2019-03-29:
Yeah, we're aware, and will have this addressed in our upcoming release. :smile: 

Also of note, this isn't actually our official support channel, so if you want these bugs to be actually noticed, you need to contact us via coral-support@google.com. I'm here, but that's because I'm nuts. :stuck_out_tongue:

Namburger, 2020-05-24:
I guess this is no longer an issue since we've moved on to apt package and better installation script/library names :)

closed 2020-05-24 by Namburger