f0cal/google-coral #2: `screen /dev/ttyUSB0 115200` fails with `[screen is terminating]`

brianthelion, 2019-03-15:
Host OS: Ubuntu 18.04
Host python3 version: 3.6.7

When running the Getting Started tutorial, I everything went smoothly up to:

```bash
$ dmesg | grep ttyUSB
[11567.500409] usb 1-2: cp210x converter now attached to ttyUSB0
[11567.503236] usb 1-2: cp210x converter now attached to ttyUSB1
[11569.062428] cp210x ttyUSB1: failed set req 0x1e size 4 status: -32
[11569.062443] cp210x ttyUSB1: failed to set baud rate to 300
```
and then
```bash
$ screen /dev/ttyUSB0 115200
[screen is terminating]
```

brianthelion, 2019-03-15:
Not sure what's up with the `dmesg` output but I think it's orthogonal to the `screen` issue. The `screen` issue I believe is due to `udev` permissions. It's quickly solved with:
```bash
$ sudo screen /dev/ttyUSB0 115200
```

hughpyle, 2019-03-16:
Alternatively: the /dev/ttyUSB0 should be group-writeable bu 'dialout', so add your user to the 'dialout' group:  
```
sudo usermod -a -G dialout $USER
```
then you'll be able to run screen without sudo.

taunusflieger, 2019-03-17:
Are you trying to connect from within a VM? Got the same when using a Parallel VM. I ended up using a Raspberry PI as the Linux box for installation

brianthelion, 2019-03-17:
No, I was connecting from a Thinkpad running Ubuntu.

asnare, 2019-03-19:
When getting started there are two USB devices that are used:

 - The first is a UART bridge that you use get the serial console. This is normally `/dev/ttyUSB0` after you plug it in.
 - The second is the OTG/data port where the firmware gets uploaded. The `udev` rule in the documentation (covering device with vendor id `0525`) ensures this gets the right permissions for `fastboot` to do its thing.

Unfortunately there's no `udev` rule for the first (terminal) device (vendor id `10c4`) which means that screen probably can't open it.

I fixed this by adding an additional rule to cover the first device, after which my `/etc/udev/65-edgetpu-board.rules` file looked like this:
```
SUBSYSTEMS=="usb",ATTRS{idVendor}=="10c4",MODE="0664",GROUP="plugdev",TAG+="uaccess"
SUBSYSTEMS=="usb",ATTRS{idVendor}=="0525",MODE="0664",GROUP="plugdev",TAG+="uaccess"
```
Note that there's also a difference from the docs here in that these lines start with SUBSYSTEM**S** instead of SUBSYSTEM.

With this minor change the rest of the QuickStart worked for me.

Setup: Debian 9.8 (VM) on a macOS Host (Virtualised via VMWare Fusion with USB 3.0 passthrough)

racky1, 2019-03-22:
So I had this problem when using an Ubuntu box so I used a Raspberry Pi 3 and was able to flash the unit with the instructions.

BarryLYH, 2019-03-27:
> Alternatively: the /dev/ttyUSB0 should be group-writeable by 'dialout', so add your user to the 'dialout' group:
> 
> ```
> sudo usermod -a -G dialout $USER
> ```
> then you'll be able to run screen without sudo.

I tried it and now I can connect to ttyUSB0. However, when I went to step "fastboot devices", I cannot find the ttyUSB1. Do you have any idea of that?

Namburger, 2020-05-24:
@BarryLYH you shouldn't need both `/dev/ttyUSB0` and `/dev/ttyUSB1`, could you clarify?

closed 2020-05-24 by Namburger

TruckDynasty, 2022-06-25:
For anyone winding up here from a search engine: you generally need to add your user to both 'dialout' *and* 'tty'

phnjau, 2024-01-11:
My problem is that i get into the serial port and it gives me a bunch of question marks and I can't  even type anything and what supposed to do is type "AT".Do you think any of you guys can give me some tips?

Thanks 

              GreaseMonkeysCode2

SantoshGovindaraj, 2024-07-17:
I had similar issue. I solved it by checking permissions of the port.
```
sudo usermod -aG dialout $USER
sudo chmod 666 /dev/ttyUSB0
```

phnjau, 2024-08-31:
Ok thanks ill try that.