f0cal/google-coral #3: Can't compile OpenCV from source due to memory errors

brianthelion, 2019-03-16:
# What I ran

```bash
$ sudo apt-get install python3-dev python3-venv
$ python -m venv _venv35
$ source _venv35/bin/activate
$ pip install --upgrade pip
$ pip install conan
$ conan install opencv/4.0.1@conan/stable --build=misisng
```

# What the docs said should happen

Build completes.

# What actually happened

Build fails; out of memory.

brianthelion, 2019-03-20:
Added a swap file. See #5 . Promptly ran out out disk space about 2/3rds of the way through the build.

CharlesCCC, 2019-03-30:
why would you use OpenCV when this thing is built for TensorFlow ???

brianthelion, 2019-03-31:
@CharlesCCC Not all computer vision is neural networks.

CharlesCCC, 2019-04-01:
okay, I got that. I just wasn't sure what's the reason behind it ? Like, what would be a good reason using Computer Vision without neural networks ?(if there other limitation ?)  

In addition, if you would like to use computer vision without neural networks(Not necessary Tensorflow, but let's just say it is). why would you pick a device like EdgeTPU (which is optimized for TensorFlow) vs. other SOC devices like Jettson Nano, RaspberryPi, Orange Pi etc.

brianthelion, 2019-04-01:
@CharlesCCC All good questions. Most computer vision pipelines "in the wild" have several steps. Some of those steps make use of the CPU and some of them make use of coprocessor hardware like GPUs and TPUs. Depending on the application, the CPU steps can just as easily be the throughput bottleneck.

We're interested in providing engineers with clear decision criteria like, "When you want to do X, use device Y." Part of that process is benchmarking the hardware against common computer vision algorithms, but CPU-bound and coprocessor-bound implementations. OpenCV has the biggest selection of such implementations all in one place.

See https://app.f0cal.com/benchmarks

maiermic, 2019-05-25:
I have not tried it myself, but [this](https://medium.com/@balaji_85683/installing-opencv-4-0-on-google-coral-dev-board-5c3a69d7f52f) guide explains how to install OpenCV 4.0 from source on Google Coral Dev Board.

Namburger, 2020-05-21:
Hi guys - 

With Mendel Day (4.0) you can install opencv as easy as this:
```
sudo apt install python3-opencv
```
If you are using an older OS, I sugggest [flashing a new image](https://coral.ai/docs/dev-board/reflash/#flash-a-new-system-image).

Cheers!

closed 2020-05-24 by Namburger