f0cal/google-coral #5: Adding a swapfile

brianthelion, 2019-03-20:
Due to lack of device RAM (see #3 ), it's important to be able to use the non-volatile memory (ie, the "disk space") for storing RAM overruns. This is what [swap space][1] is for.

Older *nix systems tended to have an entire partition dedicated to this. Modern systems tend to use a file instead.

This tutorial describes how to add a swap file to the Coral dev board.

[1]: https://en.wikipedia.org/wiki/Paging#Unix_and_Unix-like_systems

brianthelion, 2019-03-20:
There's a standard package that will handle all the mechanics of setting up the swap file for you:

```bash
$ sudo apt-get install phys-swapfile
```
Note that if the swap file grows unbounded, it can eat up all the free space on the disk that's mounting the root partition. This is the 16GB eMMC by default. If you bound the size of the swap file, though, you may be back in "out of memory" land. Picking the right size for the swap file is a bit fuzzy.

All the settings related to the swap file can be controlled by editing `/etc/dphys-swapfile`.

lleonid, 2019-03-28:
This is going to quickly exhaust write cycles of emmc device.

brianthelion, 2019-03-29:
@lleonid I'll talk about moving the swapfiled to a filesystem location that's mounted on the SD card in #6 . Working on that today.

jtgans, 2019-03-29:
Should just be a matter of doing the following:

```
$ sudo mkswap -L sdswap /dev/mmcblk1
$ sudo swapon /dev/mmcblk1
```

That will use the full SD card just for swap for that session. To make it permanent:

```
$ echo /dev/mmcblk1 none swap sw defaults 0 0 |sudo tee -a /etc/fstab
```

Note that if you don't have that exact SD card inserted, it'll fail to boot.

jtgans, 2019-03-29:
Note that SD cards and USB flash drives will also have a limited number of write cycles as well, depending on the grade and quality of the card/flash drive.

jtgans, 2019-03-29:
Also, if you want swap on an sdcard with a filesystem (say mounted at /home like in #6), do this:

```
$ sudo truncate -s2G /home/swapfile
$ sudo mkswap /home/swapfile
$ sudo swapon /home/swapfile
```

And again, to make it permanent:

```
$ echo /home/swapfile none swap sw defaults 0 0 |sudo tee -a /etc/fstab
```

pliablepixels, 2019-12-13:
A lot of issues here are marked as tutorial, but I don't see instructions in them. Are they work in progress?

I just installed a 128GB SD card in my coral board. 
I am looking to achieve 2 things:

a) A permanent 1GB swap file in the SD card
b) "Extend" my '/home/mendel" directory to also include space from the SD card whilst not losing any existing data 

Is that possible? Would someone mind posting exact instructions?

closed 2020-05-24 by Namburger