f0cal/google-coral #6: Using an SD card for extra disk space

brianthelion, 2019-03-20:
The Coral dev board comes with 16GB of non-volatile storage in the form of an eMMC. This isn't very much. If you want to expand the device's storage, you can add disk space via either an SD card or the USB3 slot.

If you want the additional storage to be "transparent" -- ie, so you don't have to change the way you do things on the device in order to use it -- then you should mount a part of the filesystem on it that's (1) frequently responsible for taking up disk space and for which (2) super fast access isn't super important. For me, `/home` fit the bill.

The rest of the tutorial describes how to format an SD card as ext4, find its device ID, and mount `/home` to that device without losing the existing home directories (if any).

ksachdeva, 2019-04-27:
> The Coral dev board comes with 16GB of non-volatile storage in the form of an eMMC. 

I think there is a typo here. The board comes with 8GB of eMMC.

I have verified it by running `df -h` on my board and even the datasheet says so -

https://coral.withgoogle.com/docs/dev-board/datasheet/#features

##

I see the reference to 16GB eMMC in another issue as well.

brianthelion, 2019-04-27:
@ksachdeva You are correct. Updated.

JVTEAM, 2019-06-03:
Is the tutorial on here?

pliablepixels, 2019-12-13:
> The rest of the tutorial describes how to format an SD card as ext4, find its device ID, and mount /home to that device without losing the existing home directories (if any).

@brianthelion would you mind posting instructions for this?

closed 2020-05-24 by Namburger