f0cal/google-coral #7: getting access to _edgetpu_cpp_wrapper_arm64.so

eejlny, 2019-03-21:
Hello,

Would someone with access to the dev board share the _edgetpu_cpp_wrapper_arm64.so. I only have access to the USB device and I am trying to use it with an ARM64 board but this file is missing and only the 32bit version is present so I cannot use it. The docs say that this is part of the dev board file system but I do not have a a dev board only the USB coral stick.

[image]

Thanks,

brianthelion, 2019-03-21:
Don't have the dev board in front of me today but can track it down for you soon-ish.

eejlny, 2019-03-22:
Thanks,

I also contacted the Google Coral team and they said they forgot and that they will add it in a future release but they did not say when.

brianthelion, 2019-03-25:
@eejlny Sorry for the late reply on this. Here's what I've got:

```bash
$ locate edgetpu | fgrep .so
/usr/lib/aarch64-linux-gnu/libedgetpu.so
/usr/lib/aarch64-linux-gnu/libedgetpu_arm64.so
/usr/lib/python3/dist-packages/edgetpu/swig/_edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so
```
The name doesn't match exactly, but I'm guessing you're talking about the latter? Those appear to come from some .debs installed by the Mendel image:

```bash
$ locate edgetpu | fgrep .deb
/var/cache/apt/archives/libedgetpu_5-2_arm64.deb
/var/cache/apt/archives/python3-edgetpu_5-2_arm64.deb
```

Let me know what you need me to send.

eejlny, 2019-03-26:
Thanks for looking into this,

yes, it is the later _edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so

Not sure if it will work but it is worth a try. Thanks again.

eejlny, 2019-03-28:
Hi brianthelion,

Please, if you get the chance you can send me the .so ?

Thanks,

blitzvb, 2019-03-29:
Can you post it here ? Thx

k2kdisk, 2019-03-29:
You can get _edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so from the dev board flash image.
```
wget https://dl.google.com/aiyprojects/mendel/enterprise/mendel-enterprise-beaker-22.zip
unzip mendel-enterprise-beaker-22.zip && cd mendel-enterprise-beaker-22
simg2img rootfs_arm64.img rootfs_arm64_ext4.img
mkdir work
sudo mount -t ext4 -o loop rootfs_arm64_ext4.img ./work
```
I try to run the stick on an aarch64 board using the .so file, but it failed with following error.
```
ERROR: Failed to retrieve TPU context.
ERROR: Node number 0 (edgetpu-custom-op) failed to prepare.

Failed in Tensor allocation, status_code: 1
Aborted
```

brianthelion, 2019-03-29:
@eejlny @blitzvb Sorry for the delay here; I had a bunch going on this week.

I was about to send you a Google Drive link, but @k2disk just gave us a better approach.

blitzvb, 2019-03-29:
@k2kdisk thanks! I appreciate.

After modifying the install.sh and cp all the required libs, I got the exact same error as you got :/

FYI, it's "running" on a jetson tx1

any ideas ?

eejlny, 2019-03-29:
Thanks,

Pity is not working.

How did you manage to get it working on the Jetson tx1. Use a 32-bit OS ?

Reader in Adaptive and Energy Efficient Computing
Fellow of the Higher Education Academy
Advanced Microelectronics and System Engineering (AMSE) MSc program director
Merchant Venturers Building
Department of Electronic and Electrical engineering
University of Bristol
phone : 0117 3315128
Fax : 0117 9545205
web: http://seis.bris.ac.uk/~eejlny/
e_mail : j.l.nunez-yanez@bristol.ac.uk<mailto:j.l.nunez-yanez@bristol.ac.uk>

blitzvb, 2019-03-29:
My bad. I meant I started the demos on a Jetson tx1 ;) ... this is why I quoted the “running” ;)

eejlny, 2019-04-03:
OK, I tried as well and I can confirm the problem:

:~/coral/python-tflite-source/edgetpu# python3 demo/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --label test_data/inat_bird_labels.txt --image test_data/parrot.jpg                  Failed to retrieve TPU context.
Node number 0 (edgetpu-custom-op) failed to prepare.

Failed in Tensor allocation, status_code: 1
Aborted (core dumped)

I tried the adding user to devgroup as in the instructions but it did not help. 

In my case dmesg does not show the coral device when I connect with usb 3 cable and usb 2 cable shows something :

[185738.070666] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[185738.219312] usb 1-1: New USB device found, idVendor=1a6e, idProduct=089a
[185738.219317] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber= 

I guess it will be good to check if the USB device is recognised correctly in your systems.

blitzvb, 2019-04-03:
try lsusb. For me the USB device was seen.

eejlny, 2019-04-03:
yes, I think I can see the device as usb 2.0

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 1a6e:089a Global Unichip Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Still the same error with the tpu context 😞

k2kdisk, 2019-04-03:
On my RasPi3, the usb stick works fine and lsusb is same as @eejlny 
```
Bus 001 Device 004: ID 1a6e:089a Global Unichip Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
But after running the demo, the vid:pid changed :
```
Bus 001 Device 005: ID 18d1:9302 Google Inc.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

In my imaginagion, demo script or cpp_wrapper.so setup TPU as usb gadget on arm32, but as PCIE board on arm64.
The dev board block diagram shows that TPU is connected on PCIE bus.

samsterckval, 2019-04-11:
Any updates on this?  I would also like to run it on a aarch64 board, tried the usual stuff of installing the arm64 libraries, but indeed got to the problem it isn't connected through PCIE.

blitzvb, 2019-04-11:
I heard that arm64(aarch64) will be included in their next release.

samsterckval, 2019-04-11:
Maybe a stupid question, but I don't seem to directly find anything about it : Do we have any idea when the next release is due?

PeterMalkin, 2019-04-12:
Seems to be up on coral website now :) 
https://coral.withgoogle.com/news/updates-04-2019/

samsterckval, 2019-04-12:
It does not seem to talk about the aarch64 support for the USB accelerator though, if anyone could confirm it works, would be great.  I have time this afternoon(CET) to test it else.

blitzvb, 2019-04-12:
I cannot test on my aarch64 board yet, but the SDK include the arm64 lib and the install.sh seems to take in account that architecture ;)

[image]

blitzvb, 2019-04-12:
it install properly but the inference is not working :/

`nvidia@tegra-ubuntu:/usr/local/lib/python3.5/dist-packages/edgetpu/demo$ python3 object_detection.py \
> --model ~/Downloads/mobilenet_ssd_v2_face_quant_postprocess_edgetpu.tflite \
> --input ~/Downloads/face.jpg \
> --output ~/detection_results.jpg
Traceback (most recent call last):
  File "object_detection.py", line 110, in <module>
    main()
  File "object_detection.py", line 75, in main
    engine = DetectionEngine(args.model)
  File "/usr/local/lib/python3.5/dist-packages/edgetpu/detection/engine.py", line 55, in __init__
    super().__init__(model_path)
  File "/usr/local/lib/python3.5/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 300, in __init__
    this = _edgetpu_cpp_wrapper.new_BasicEngine(*args)
RuntimeError: Failed to allocate tensors.`

PeterMalkin, 2019-04-12:
If you don't mind, can you send the logs and the model to the email address
listed here: https://coral.withgoogle.com/support/

PeterMalkin, 2019-04-12:
Do the default models that are listed on the coral website work?

blitzvb, 2019-04-12:
it is the default model with the default demos (I tried classification and detection). 

for the full log, I already sent it to coral-support@google.com this morning with that email : vincent@goview.co

k2kdisk, 2019-04-12:
I cleanup my arm64 board rootfs and install the new one, but not working.
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 18, in swig_import_helper
    fp, pathname, description = imp.find_module('_edgetpu_cpp_wrapper', [dirname(__file__)])
  File "/usr/lib/python3.6/imp.py", line 297, in find_module
    raise ImportError(_ERR_MSG.format(name), name=name)
ImportError: No module named '_edgetpu_cpp_wrapper'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "classify_image.py", line 19, in <module>
    from edgetpu.classification.engine import ClassificationEngine
  File "/usr/local/lib/python3.6/dist-packages/edgetpu/classification/engine.py", line 17, in <module>
    from edgetpu.basic.basic_engine import BasicEngine
  File "/usr/local/lib/python3.6/dist-packages/edgetpu/basic/basic_engine.py", line 15, in <module>
    from edgetpu.swig.edgetpu_cpp_wrapper import BasicEngine
  File "/usr/local/lib/python3.6/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 28, in <module>
    _edgetpu_cpp_wrapper = swig_import_helper()
  File "/usr/local/lib/python3.6/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 20, in swig_import_helper
    import _edgetpu_cpp_wrapper
ModuleNotFoundError: No module named '_edgetpu_cpp_wrapper'
```

The new installer not seems to be include _edgetpu_cpp_wrapper. Where is it?

samsterckval, 2019-04-12:
It seems the install.sh file does not rename the .so file.
you can copy the aarch64 file and rename it to _edgetpu_cpp_wrapper.so, then all works fine for me!  I have a speedup of about 3 compared to a 32-bit USB2.0 system.

k2kdisk, 2019-04-12:
@samsterckval Thanks! I can run a sample on aarch64 board.
```
$ cd /usr/local/lib/python3.6/dist-packages/edgetpu/swig/
$ sudo cp _edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so _edgetpu_cpp_wrapper.so
$ python3 /usr/local/lib/python3.6/dist-packages/edgetpu/demo/classify_image.py --model ~/Downloads/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --label ~/Downloads/inat_bird_labels.txt --image ~/Downloads/parrot.jpg
W0412 13:02:03.809260   16731 package_registry.cc:65] Minimum runtime version required by package (5) is lower than expected (10).
---------------------------
Ara macao (Scarlet Macaw)
Score :  0.761719
```

blitzvb, 2019-04-12:
For me, on a Jetson TX1, it was a different bug. here what I did to make it work:

change install.sh like that:

`  sudo -H python3 -m pip install --no-deps "${WHEEL}"`

run any app that use the coral API with sudo:

`sudo python3 classify_image.py --model ~/devel/ai/demo/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --label ~/devel/ai/demo/inat_bird_labels.txt --image ~/devel/ai/demo/parrot.jpg`

otherwise I get that error:

`RuntimeError: Failed to allocate tensors.`

Note to others, copying the .so did change anything on my system... and it works without it.

nitschw, 2019-06-17:
Hi everyone, @k2kdisk @samsterckval @eejlny I was about to try to get the parrot classification example running on the Coral USB accelerator with a Jetson Nano but am getting this same error (no module _edgetpu_cpp_wrapper_arm64.so).

I was just wondering if you ever got this fix process totally nailed down for arm64? It appears you had some success, but it is a little bit unclear from your above conversation exactly how to correct the issue.

eejlny, 2019-06-17:
The updated version contains the necessary files.

Check in the script and the folders to make sure that the names match since they do change slightly

jveitchmichaelis, 2019-06-19:
For what it's worth, I'm running the USB accelerator on Arm64. I had no issue with the `install.sh` script, but also you can build from scratch (e.g. if you wanted to use an Anaconda environment).

Clone the repository: https://coral.googlesource.com/edgetpu/+/refs/heads/release-chef

Run:

```
cd edgetpu
./build_package.sh
python setup.py sdist bdist_wheel

cd dist
pip install <edgetpu_version>-py3-none-any.whl
```

Namburger, 2020-05-24:
Hi guys, Nam from the google-coral team here. Apologies for this issue but I guess this is no longer applicable because the way we build out api changes significantly since this issue open. To clarify, there are 2 so for an average user of out API:

1) The `libedgetpu.so` which allows you to interact with the edgetpu
2) The swig wrapper so that binds the python interface with our C++ API. This .so naming location and installation path is the main reason of this issue. We have evolved this to use a more generic naming and installation path as well as apt package to make this less of an issue. 

Closing this for now, feel free to ask more questions regarding this if something isn't clear!

closed 2020-05-24 by Namburger