f0cal/google-coral #8: Spoofing the Linux distro and version

brianthelion, 2019-03-22:
One of the annoying things about the dev board is that, while it appears to run a fairly vanilla version of Debian Linux, it *reports* that it runs Mendel Linux.

This is problematic because some of the Debian tools and packages that would otherwise be compatible with the dev board may balk at being operated on Mendel.

Fortunately there is a workaround for at least some such cases:

The standard way of checking the Linux distro name and version number is a little executable called `lsb_release`, which itself pulls from a configuration file in `/etc`.

It's possible to get `lsb_release` to report something other than "Mendel Linux" using [this slick little `lsb_release` hack][1].

[1]: https://www.hiroom2.com/2017/02/21/linux-change-reference-file-of-lsb-release-from-etc-lsb-release/

jtgans, 2019-03-28:
Which tools are you attempting to use that cause you problems?

brianthelion, 2019-03-29:
@jtgans https://github.com/saltstack/salt-bootstrap

jtgans, 2019-03-29:
Perhaps it would be better to send a PR to salt-bootstrap to support Mendel? Mendel is a proper derivative of Debian stretch (stable), and we're following the guidelines set forth here: https://wiki.debian.org/Derivatives/Guidelines, which states we can't actually use the name Debian.

brianthelion, 2019-03-29:
@jtgans Noted. I'll ping them on Slack.

Namburger, 2020-05-24:
FYI, Mendel Day (4.0) is a derivative of Buster, I've found that most packages can be installed if you give the `/etc/sources.list.d` files the correct source of arm64 and buster, similar to this answer on docker: https://github.com/f0cal/google-coral/issues/32#issuecomment-571629174

Another way is downloading release binaries that were compiled for `arm64`, which is not very commonly supported, unfortunately. But I got that working for containers like kubenettes.

I've never used `salt-bootstrap` before but it looks like their [releases](https://github.com/saltstack/salt-bootstrap/releases) are only for x86_64 platform which is pretty standard. Maybe compiling from source? 
Anyhow, for thirdparty software/libraries like this, we usually recommend looking through their forum for supports with arm64/debian platforms!

closed 2020-05-24 by Namburger