f0cal/google-coral #10: Updating the Python version

brianthelion, 2019-04-11:
One of the difficult parts of working with the devboard is the Python version (3.5) that we've inherited from Debian Stretch. This is a pre-`orderedict`-everywhere version, which can create a lot of confusion if you're used to `kwargs` being ordered.

Unfortunately, Debian hasn't provided Stretch backports for newer Python versions. In fact, They've taken `python3.6` out of their [pool](http://ftp.debian.org/debian/pool/main/p/) altogether. Not sure what that's about.

brianthelion, 2019-04-11:
It might be possible to leverage Ubuntu's excellent `ppa:deadsnakes/ppa`, but I decided to try a different tack after discovering a `python3.6` [backport script](https://github.com/chriskuehl/python3.6-debian-stretch) on GitHub.

The results is https://github.com/f0cal/python3.7-debs. Follow the instructions there and you're good to go with `python3.7`!

adr-arroyo, 2019-06-11:
Hello @brianthelion, thanks for your contribution on the script.
I installed by source Python 3.7 in the coral, but I cannot install pandas lib (with   pip3 install pandas), but I can install other libraries as numpy. Do you have this problem also?

Regards

brianthelion, 2019-06-12:
@adr-arroyo Can you post some logs of how you're trying to run the install and what the output looks like?

adr-arroyo, 2019-06-17:
Hello,
I actually did not have any logs, pip just froze and couldnt continue. Let me try again and see if I have some output logs.

adr-arroyo, 2019-06-27:
Hello,

Indeed I was able to get pandas on the coral, the problem was that it was taking sooo long to download that it appeared to be frozen.

If you run `pip` with option `--vvv` you will see what the installation is doing on the console.

Cheers!

Namburger, 2020-05-24:
Thanks! We're now onto python3.7 with Mendel 4.0 being a derivative of buster :)

closed 2020-05-24 by Namburger

Rhyzhang, 2022-11-13:
Hi @brianthelion ,

It seems `sudo add-apt-repository ppa:deadsnakes/ppa` does not work as I get this error:
```
WARNING:root:could not open file '/etc/apt/sources.list'

WARNING:root:could not open file '/etc/apt/sources.list'

Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 93, in get_sources
    (self.id, self.codename))
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Mendel/eagle
```

Currently, Python3 on Mendel is running on Python 3.7.3 however matplotlib needs Python 3.8 in order to install. Do you know how I can update Python 3.7.3 to Python 3.8? 

Best,
Ryan