f0cal/google-coral #11: ssh Permission denied (publickey).

SX-GitHub, 2019-04-21:
* Doc you were trying to follow: https://coral.withgoogle.com/docs/dev-board/reflash/
* Your host OS: Debian ARM64 stretch
* Your Python3 version: N/A

# ssh mendel@192.168.100.2

# should show password:

# I got error "ssh Permission denied (publickey)."
-- I was able to ssh to 2.0 Beaker minutes ago before flash and reboot!

[coral_board_ssh_log.txt](https://github.com/f0cal/google-coral/files/3101589/coral_board_ssh_log.txt)

SX-GitHub, 2019-04-22:
OK the answer is in the bottom part of the page https://coral.withgoogle.com/docs/dev-board/mdt/#connect-to-a-boards-shell

closed 2019-04-22 by SX-GitHub

prasad456, 2019-08-16:
> OK the answer is in the bottom part of the page https://coral.withgoogle.com/docs/dev-board/mdt/#connect-to-a-boards-shell

apt-get not working for mac os?