f0cal/google-coral #12: USB Accelerator - C++ API - Segmentation fault

JaviBonilla, 2019-04-25:
* Doc you were trying to follow: 
https://coral.withgoogle.com/docs/edgetpu/api-cpp/
https://coral.googlesource.com/edgetpu-native/+/refs/heads/release-chef/libedgetpu/edgetpu.h
* Your host OS: _openSUSE Tumbleweed x86_64 (it is unsupported but it is working fine)_
* Your Python3 version: _Not Python, I am using C++ API and the USB Accelerator_

# What I ran
Everything is working great with respect to the C++ API besides:

- edgetpu::EdgeTpuManager::GetSingleton()->EnumerateEdgeTpu()
- edgetpu::EdgeTpuManager::GetSingleton()->Version()

# What the docs said should happen

This is not in the docs but in a header file (https://coral.googlesource.com/edgetpu-native/+/refs/heads/release-chef/libedgetpu/edgetpu.h).

- List of TPUs
- String defining the version.

# What actually happened

I am getting segmentation fault, maybe those functions are missing in the distributed dynamic libraries (*.so), I tested this with the _libedgetpu_arm64.so_ library.

brianthelion, 2019-04-26:
@JaviBonilla If those functions were missing from the `.so` you would see a linker errors when you try to compile your application code, as opposed to a segmentation fault.

JaviBonilla, 2019-04-26:
@brianthelion yes, you are right! I checked again and the _EnumerateEdgeTpu_ method is in the docs and in an example.

https://coral.withgoogle.com/docs/edgetpu/multiple-edgetpu/
https://coral.googlesource.com/edgetpu-native/+/refs/heads/release-chef/edgetpu/cpp/examples/two_models_two_tpus_threaded.cc

brianthelion, 2019-04-27:
@JaviBonilla Did you try running your application in a `gdb` sandbox to get a backtrace after the segfault?

JaviBonilla, 2019-04-29:
Ok, so when I execute the following code,
               
```
// Get vector of TPU objects
const std::vector<edgetpu::EdgeTpuManager::DeviceEnumerationRecord> &available_tpus =
     edgetpu::EdgeTpuManager::GetSingleton()->EnumerateEdgeTpu();

// Print number of TPU objects
cout << "Number of TPUs: " << available_tpus.size();

// Iterate over TPU objects and print some information
for(edgetpu::EdgeTpuManager::DeviceEnumerationRecord edgetpu : available_tpus)
     cout << "TPU:" << (edgetpu.type == edgetpu::DeviceType::kApexUsb ? "USB" : "PCI") 
          << edgetpu.path.c_str();
```
I get the following information.

> Number of TPUs: 3689348814741910324

I think the -3689348814741910324 in hex is 0xCCCCCCCCCCCCCCCC, and this represents a value that a local variable gets when it is not initialized in debug mode.

The segmentation fault happens when accessing the first element of the `available_tpus` vector, that is the `edgetpu` variable in the for loop.

brianthelion, 2019-04-29:
@JaviBonilla  Can you share the gdb backtrace? So instead of 
```
Example usage:
1. Create directory /tmp/edgetpu_cpp_example
2. wget -O /tmp/edgetpu_cpp_example/inat_bird_edgetpu.tflite \
     http://storage.googleapis.com/cloud-iot-edge-pretrained-models/canned_models/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite
3. wget -O /tmp/edgetpu_cpp_example/inat_plant_edgetpu.tflite \
     http://storage.googleapis.com/cloud-iot-edge-pretrained-models/canned_models/mobilenet_v2_1.0_224_inat_plant_quant_edgetpu.tflite
4. wget -O /tmp/edgetpu_cpp_example/bird.jpg \
     https://farm3.staticflickr.com/4003/4510152748_b43c1da3e6_o.jpg
5. wget -O /tmp/edgetpu_cpp_example/plant.jpg \
     https://c2.staticflickr.com/1/62/184682050_db90d84573_o.jpg
6. cd /tmp/edgetpu_cpp_example && mogrify -format bmp *.jpg
7. Build and run `two_models_two_tpus_threaded`
```
when you get to step (7) run `gdb two_models_two_tpus_threaded`. That will open the `gdb` prompt. Type `r` for "run". When you see the segfault, type `bt` for backtrace` and post the result.

JaviBonilla, 2019-04-30:
@brianthelion Sure, this is the backtrace.

```
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:304
#1  0x00007ffff7d8760a in std::char_traits<char>::copy (__n=26, __s2=0x21 <error: Cannot access memory at address 0x21>, __s1=<optimized out>)
    at /usr/src/debug/gcc8-8.3.1+r269200-1.1.x86_64/obj-x86_64-suse-linux/x86_64-suse-linux/libstdc++-v3/include/bits/char_traits.h:352
#2  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_S_copy (__n=26, __s=0x21 <error: Cannot access memory at address 0x21>, __d=<optimized out>)
    at /usr/src/debug/gcc8-8.3.1+r269200-1.1.x86_64/obj-x86_64-suse-linux/x86_64-suse-linux/libstdc++-v3/include/bits/basic_string.h:340
#3  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_S_copy_chars (__k2=<optimized out>, __k1=0x21 <error: Cannot access memory at address 0x21>, __p=<optimized out>)
    at /usr/src/debug/gcc8-8.3.1+r269200-1.1.x86_64/obj-x86_64-suse-linux/x86_64-suse-linux/libstdc++-v3/include/bits/basic_string.h:382
#4  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*> (this=0x7fffffffd388, __beg=0x21 <error: Cannot access memory at address 0x21>, __end=<optimized out>)
    at /usr/src/debug/gcc8-8.3.1+r269200-1.1.x86_64/obj-x86_64-suse-linux/x86_64-suse-linux/libstdc++-v3/include/bits/basic_string.tcc:225
#5  0x0000000000410ded in edgetpu::EdgeTpuManager::DeviceEnumerationRecord::DeviceEnumerationRecord(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&) ()
#6  0x0000000000410e13 in std::_Head_base<1ul, edgetpu::EdgeTpuManager::DeviceEnumerationRecord, false>::_Head_base(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&) ()
#7  0x0000000000410613 in std::_Tuple_impl<1ul, edgetpu::EdgeTpuManager::DeviceEnumerationRecord, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::_Tuple_impl(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#8  0x000000000040f114 in std::_Tuple_impl<0ul, main::{lambda(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}, edgetpu::EdgeTpuManager::DeviceEnumerationRecord, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::_Tuple_impl<{lambda(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}&, edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void>({lambda(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}&, edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#9  0x000000000040f0a6 in std::tuple<main::{lambda(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}, edgetpu::EdgeTpuManager::DeviceEnumerationRecord, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::tuple<{lambda(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}&, edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, true>({lambda(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}&, edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#10 0x000000000040f1b3 in std::thread::_Invoker<std::tuple<std::decay<main::{lambda(edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char [... cut ...] geTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}&, edgetpu::EdgeTpuManager::DeviceEnumerationRecord const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#12 0x000000000040ebb6 in main ()
```

larrylart, 2019-05-05:
I am getting the same, segmentation fault, while calling EnumerateEdgeTpu on Ubuntu 18.04/Armv7. 
I added a basic c++ sample of mobilenet ssd v2 at https://github.com/larrylart/codrive/tree/master/hw_sampling/google_coral, which works fine otherwise, however it fails if you try enumerate/use multiple devices. 
@JaviBonilla it does look like an initialisation bug in the library.

JaviBonilla, 2019-05-05:
Thanks for the info and the code @larrylart. Yes, the bug seems to be that the list of devices is not properly initialized and it reports a wrong number of TPU devices.

I didn't inform about this to the Coral team because I am on an unsupported OS, but since you tested this on Ubuntu perhaps we should let them know at https://coral.withgoogle.com/support/.

larrylart, 2019-05-05:
I get the same segmentation fault on the coral dev board as well, that with their own Mendel debian flavour, so it's definitely due to a bug in the libedgetpu. 
I noticed that you can set SetVerbosity(10) and get  debug info from the lib, however I could not find anything that might help, just that the standard NewEdgeTpuContext() does also make a call internally to enumerate devices, obviously not crashing.

JaviBonilla, 2019-05-07:
@larrylart Thanks for the additional tests! I have just written to them and also sent a link to this discussion so they can have more info.

larrylart, 2019-05-07:
@JaviBonilla, well, as of few hours ago, that I did as well and also sent a link to this thread ... I guess the more the merry. So, far I   got a confirmation that their engineers are looking after, fingers crossed.

weiranzhao, 2019-05-07:
Folks, do you see segmentation fault if you compile the binary with given tool-chain using the docker image? My hypothesis now is that this is caused by linking against libstdc++ instead of libc++.

libedgetpu.so is compiled against libc++, and I see from the stacktrace above, mention of libstdc++
such as x86_64-suse-linux/libstdc++-v3/include/bits/basic_string.h

lleonid, 2019-05-07:
alternatively on coral dev board you could:
sudo apt-get install clang edgetpu-dev libc++-dev
and then:
mendel@silly-apple:~$ cat test.cc
#include <iostream>
#include <edgetpu.h>

int main(void) {
        auto v = edgetpu::EdgeTpuManager::GetSingleton()->Version();
        std::cout << v << std::endl;
}
mendel@silly-apple:~$ clang++ -std=c++11 -stdlib=libc++ -ledgetpu -I/home/mendel/edgetpu-native/tensorflow -lsupc++ test.cc
mendel@silly-apple:~$ ./a.out
BuildLabel(N/A), RuntimeVersion(10)
mendel@silly-apple:~$

JaviBonilla, 2019-05-07:
Thanks @weiranzhao, ok! so that's probably the problem. I switched to clang and libc++ from gcc and libstdc++ to compile this example because it is easier to use libc++ in clang, but I still have libstdc++ dependencies in the TensorFlow Lite library because I compiled it with make and gcc.

I will try to compile TensorFlow Lite without libstdc++ dependencies and test the example again. I will let you know when I test this.

larrylart, 2019-05-07:
Getting tensorflow with clang is easy enough ... however compiling opencv with clang and neon is another story. @JaviBonilla let us know if this does it for you and/or if you managed to also compile opencv with clang as well.

brianthelion, 2019-05-07:
@larrylart The recent `conan` OpenCV recipes should get you there with very few headaches. If they don't, let me know and I'll post one.

JaviBonilla, 2019-05-07:
@larrylart I do not use OpenCV in the examples I am going to test first, I use only C++, Qt/QML and TensorFlow or TensorFlow Lite. 

I have some examples for desktop, iOS, Android and Raspberry Pi. For example this for Raspberry Pi:  https://mechatronicsblog.com/raspberry-pi-tensorflow-lite-and-qt-qml-object-detection-example/.

Besides to compile TensorFlow Lite with clang, I have to figure it out how to use clang with qmake.

Or maybe it is easier to use gcc with libc++, some info is given at https://libcxx.llvm.org/docs/UsingLibcxx.html. This maybe also easier for OpenCV as well.

JaviBonilla, 2019-05-08:
@larrylart @weiranzhao Ok, so I confirm that the `two_models_two_tpus_threaded.cc` example works if it is linked to `libc++` instead of `libstdc++`.

larrylart, 2019-05-08:
Thank you @weiranzhao @JaviBonilla - yes the issue is due to the fact that the api is linked against libc++.
I am still in struggle to get opencv 4.1 to link exclusively against libc++, and so far has been proven extremely stubborn - after 10-20 attempts it still links both libcc++ and stdlib for some reason ignoring the flags to force libc++ only. 

It will be great if Coral team will release an api version linked against stdlib, as most of the projects are release/setup with stdlib - it will save us many days of attempts to align other libraries with libc++

larrylart, 2019-05-21:
Anyone here managed to build opencv with libstdc++ and/or link opencv with edgetpu library without segmentation fault when using mutiple tpus query ?

After spending two full days trying to get opencv to compile libc++ and running in an exponential number of dependencies I had to compile to get opencv to compile with libc++, I give up for now. I play with one tpu at the time for time being ... I hope Coral team sees reason here, as libc++ is enforced by apple, however Apple has no SBCs / edge platforms - and all others come with libraries built by default with libstdc++, so unless you are a developer which happen to own an apple and don't care much abut deploying it on an edge platform and/or integrating other processing frameworks ... well you are alittle bit stuck.

Namburger, 2020-05-21:
Hi all, too many things going on here haha

Okay so regarding tflite build issue, there are too many things to talk about, but this issue can give a good gerneral summary: https://github.com/google-coral/edgetpu/issues/78

Regarding building opencv, you can just checkout my repo :)
https://github.com/Namburger/edgetpu-detection-camera

My hope is that all these issues be gone after we open source the library :)

Cheers

closed 2020-05-24 by Namburger