f0cal/google-coral #13: ModuleNotFoundError: No module named '_edgetpu_cpp_wrapper' on ubuntu 19.04 with python 3.7

ckuethe, 2019-04-25:
* Doc you were trying to follow: https://coral.withgoogle.com/docs/accelerator/get-started/
* Your host OS: Ubuntu 19.04 x86_64
* Your Python3 version: 3.7.3

# What I ran
```
$ python3 classify_image.py \
--model ~/Downloads/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
--label ~/Downloads/inat_bird_labels.txt \
--image ~/Downloads/parrot.jpg
```

# What the docs said should happen
```
Ara macao (Scarlet Macaw)
Score :  0.761719
```

# What actually happened
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 18, in swig_import_helper
    fp, pathname, description = imp.find_module('_edgetpu_cpp_wrapper', [dirname(__file__)])
  File "/usr/lib/python3.7/imp.py", line 296, in find_module
    raise ImportError(_ERR_MSG.format(name), name=name)
ImportError: No module named '_edgetpu_cpp_wrapper'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "classify_image.py", line 19, in <module>
    from edgetpu.classification.engine import ClassificationEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/classification/engine.py", line 17, in <module>
    from edgetpu.basic.basic_engine import BasicEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/basic/basic_engine.py", line 15, in <module>
    from edgetpu.swig.edgetpu_cpp_wrapper import BasicEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 28, in <module>
    _edgetpu_cpp_wrapper = swig_import_helper()
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 20, in swig_import_helper
    import _edgetpu_cpp_wrapper
ModuleNotFoundError: No module named '_edgetpu_cpp_wrapper'
```

I'm guessing this is because there is no `37m-x86_64` build...
```
$ find /usr/local/ -name \*edgetpu\*
/usr/local/lib/python3.7/dist-packages/edgetpu-1.9.2.dist-info
/usr/local/lib/python3.7/dist-packages/edgetpu
/usr/local/lib/python3.7/dist-packages/edgetpu/basic/edgetpu_utils.py
/usr/local/lib/python3.7/dist-packages/edgetpu/basic/__pycache__/edgetpu_utils.cpython-37.pyc
/usr/local/lib/python3.7/dist-packages/edgetpu/swig/_edgetpu_cpp_wrapper.cpython-35m-arm-linux-gnueabihf.so
/usr/local/lib/python3.7/dist-packages/edgetpu/swig/__pycache__/edgetpu_cpp_wrapper.cpython-37.pyc
/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py
/usr/local/lib/python3.7/dist-packages/edgetpu/swig/_edgetpu_cpp_wrapper.cpython-35m-x86_64-linux-gnu.so
/usr/local/lib/python3.7/dist-packages/edgetpu/swig/_edgetpu_cpp_wrapper.cpython-36m-x86_64-linux-gnu.so
/usr/local/lib/python3.7/dist-packages/edgetpu/swig/_edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so
```

ckuethe, 2019-04-25:
As a nasty workaround, you can do `cd /usr/local/lib/python3.7/dist-packages/edgetpu/swig/ && sudo ln -s _edgetpu_cpp_wrapper.cpython-3{6,7}m-x86_64-linux-gnu.so`

which somewhat works for the parrot demo, albeit with a warning:
`W third_party/darwinn/driver/package_registry.cc:65] Minimum runtime version required by package (5) is lower than expected (10).`

ckuethe, 2019-04-25:
installed from `edgetpu_api_1.9.2.tar.gz`

scottlamb, 2019-06-06:
Hit a similar error: my system (ODROID-XU4, running Ubuntu 18.04.2 LTS) comes with Python 3.6, and `edgetpu_api_1.9.2.tar.gz` has the arm32 version of the swig wrapper compiled only for Python 3.5.

Symlinking the file as suggested above works at least well enough to run the demo. (It initially failed with `RuntimeError: Failed to allocate tensors.` but I saw at google-coral/edgetpu-platforms#1 that this is just a permissions problem.)

Is there source available for the swig wrapper, so we can just rebuild it? I looked around a bit starting from https://coral.withgoogle.com/resources/ but didn't find it. I was a little surprised; I wouldn't expect there to be any secret sauce in this component.

ECE-Engineer, 2019-07-02:
I followed this post:
https://medium.com/@aniketvartak/hi-sam-great-article-5a0299e2f7cd

robmarkcole, 2019-07-13:
I am getting this error with raspbian Buster on  a pi 3

z0d1ac1999, 2019-07-16:
This command below fixed it for me on raspian Buster on a pi 3.  It's sort of like ECE-Engineer's solution but for arm:

sudo cp _edgetpu_cpp_wrapper.cpython-35m-arm-linux-gnueabihf.so _edgetpu_cpp_wrapper.cpython-37m-arm-linux-gnueabihf.so

Namburger, 2019-09-04:
it's possible that you just have 2 python version installation also and the one your're using just happen to not be linked to 
"/usr/local/lib/python3.6/dist-packages"

if it's not in the out put of this: 

"$ python3 -c 'print(__import__("sys").path)'" 

then you may need to use your system python instead

LudoS0, 2019-09-16:
> For now the edgetpu_cpp_wrapper.py doesn't seem to be dynamic enough to pick the right version automatically.
> 
> I got it working by renaming the .so file to _edgetpu_cpp_wrapper.so. I should note that I also had to choose the .so file that corresponded to my machine. E.g the _edgetpu_cpp_wrapper.cpython-36m-x86_64-linux-gnu.so I renamed.
> 
> I should recommend anyone going this direction to make a directory for the four available .so files.
> 
> Also if you are using "any" of nvidia's jetson boards (e.g. nano, tx1, tx2, xavier) then rename and use the aarch .so file.
> 
> Let me know if this helps. 😃

Hi @ECE-Engineer, how did you rename a file in that directory? I couldn't rightclick>rename, like it was blocked for renaming. By the way I'm using Xavier. Thanks!

powderluv, 2019-09-22:
I had python3.7 (built mendel from master branch) so I did:

cd /usr/lib/python3/dist-packages/edgetpu/swig/
sudo ln -s _edgetpu_cpp_wrapper.cpython-35m-aarch64-linux-gnu.so _edgetpu_cpp_wrapper.cpython-37m-aarch64-linux-gnu.so

hansamann, 2019-09-26:
I am on raspbian buster and followed this advice:

sudo cp _edgetpu_cpp_wrapper.cpython-35m-arm-linux-gnueabihf.so _edgetpu_cpp_wrapper.cpython-37m-arm-linux-gnueabihf.so

When I now execute my python file, the error is a different one:

root@487cac231a2a:/app# python3 mirror.py 
Traceback (most recent call last):
  File "mirror.py", line 6, in <module>
    from teachable import initialize
  File "/app/teachable.py", line 28, in <module>
    from embedding import KNNEmbeddingEngine
  File "/app/embedding.py", line 18, in <module>
    from edgetpu.basic.basic_engine import BasicEngine
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/basic/basic_engine.py", line 17, in <module>
    from edgetpu.swig.edgetpu_cpp_wrapper import BasicEnginePythonWrapper
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 28, in <module>
    _edgetpu_cpp_wrapper = swig_import_helper()
  File "/usr/local/lib/python3.7/dist-packages/edgetpu/swig/edgetpu_cpp_wrapper.py", line 24, in swig_import_helper
    _mod = imp.load_module('_edgetpu_cpp_wrapper', fp, pathname, description)
  File "/usr/lib/python3.7/imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "/usr/lib/python3.7/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: libunwind.so.8: cannot open shared object file: No such file or directory

Does anyone have any clue?

ckuethe, 2019-09-26:
The problem is `ImportError: libunwind.so.8: cannot open shared object file: No such file or directory`

You need to install libunwind. Start with `apt search unwind` to see what version is available for your machine

Namburger, 2020-05-24:
Please check this answer :)
https://github.com/f0cal/google-coral/issues/7#issuecomment-633244553

closed 2020-05-24 by Namburger