f0cal/google-coral #14: coral-mdt-stable InRelease' doesn't support architecture 'arm64'

piperod, 2019-04-26:
* Doc you were trying to follow: https://coral.withgoogle.com/docs/dev-board/mdt/
* Your host OS: linux
* Your Python3 version:3

# What I ran : 
# Add the repo:
echo "deb https://packages.cloud.google.com/apt coral-mdt-stable main" | sudo tee /etc/apt/sources.list.d/mdt.list

# Add Google keys
curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -

# Then install the package:
sudo apt-get update

sudo apt-get install mendel-development-tool

# What the docs said should happen
mdt installed 

# What actually happened
coral-mdt-stable InRelease' doesn't support architecture 'arm64'

TheCedarPrince, 2019-05-24:
Hey @piperod wanted to let you know I figured out the problem.

You can easily miss it in the instructions but you need to install this on the host computer and not on the coral board. 

I sat for a while trying to install it on the coral board and kept running into issues like yours - then I reread the directions and sure enough, you have to install mdt on the host computer.

Hope that helps!

adr-arroyo, 2019-06-25:
What about installing the mdt in an actual arm64 like raspberry-pi? 

Is it supported anyhow? For non-linux and non-mac users (windows) it makes it very hard to work with the coral.

piperod, 2019-06-25:
Sorry I forgot to give an update on this. And you were right. I totally got lost by the instructions and then I read carefully that I supposed to install it on the host computer. I ran into similar issues trying to install conda on the board and some libraries. I learned later that the board actually runs on 32 bit which make any library that runs on 64bit inadequate. I will try to publish as soon as I finish some of my path, because there is very limited information available.

adr-arroyo, 2019-06-25:
Dear @piperod 

I agree that there is very very limited information available (actually I did not know that the board runs on 32-bit, where did you find this?)

If you please comment here when and where you release your exercises, I would appreciate it :).

I have also openned an issue about installing docker on the board [here](https://github.com/f0cal/google-coral/issues/32).

Cheers!

Namburger, 2020-05-21:
Hi guys, to clarifying some things here, The board has a 64 bits arm architect!
It's the older version of the OS that we ship that only supports 32 bits. With Mendel Day (4.0) it is now aarch64 and you can find this with `uname -a`.

Mdt is a package used to connect to the board, it's just a friendly wrapper of ssh. There is no need to install it to the board, but you can definitely use ssh on it to connect to other machines!

Cheers!

closed 2020-05-24 by Namburger