f0cal/google-coral #15: Heat and power usage at idle is high

larrylart, 2019-04-29:
* Your host OS: coral dev board

# What I ran
Nothing, at idle, and while running c++ mobile net sample app

# What actually happened

The board heats up while at idle (no other devices attached and wifi off) and power usage is 2.8W, well over what the  i.MX 8M spec is, at idle - https://www.nxp.com/docs/en/nxp/application-notes/AN12118.pdf

Fan does not seem to bother much to run either, I could not find any sensor temp config. Turning on the fan manually will cool down - but power usage goes p to 3.5W 

While running inference, a small app I wrote in c++/tensorflow lite, with an usb webcamera attached brings  the power usage to 5.4 W 

Not so much different from odroid XU4 which is octo-core and know power hungry at idle:

odroid xu4 idle: 0.260 A / 3.1 W
odroid xu4 + TPU + Logitech 920 WEBCAM: 0.510 A / 6.1 W - 24.89 Real FPS / 42.90 Inference FPS (23ms) -  load average: 0.52, 0.58, 0.62
odroid xu4 + Logitech 920 WEBCAM: 0.400 A /4.9 W - load average: 0.64, 0.61, 0.62
TPU usage - 1.0W - 1.2 W ?

TPU DEV BOARD idle: 0.250 A / 2.9 W --- 0.220 A / 2.6 W
TPU DEV BOARD + Logitech 920 WEBCAM: 0.460 A / 5.4 W - 23.00 Real FPS / 76.54 Inference FPS (13ms) - load average: 0.87, 0.58, 0.26

I was expecting better performance, power wise at idle, any of you guys had the same issue with heat and power consumption?

Thanks,

Larry

brianthelion, 2019-04-29:
@larrylart Just happened to notice this buried in some [Coral source code](https://coral.googlesource.com/edgetpu-native/+/refs/heads/release-chef/edgetpu/cpp/examples/two_models_two_tpus_threaded.cc). Might be useful, might not.

```
// To reduce variation between different runs, one can disable CPU scaling with
//   sudo cpupower frequency-set --governor performance
```

lleonid, 2019-04-30:
@brianthelion That will only increase power consumtion by forcing CPU cores to operate at higher frequency.

@larrylart You can change fan trip temperature via /sys/class/thermal/thermal_zone0/trip_point_4_temp

zorrobyte, 2019-09-27:
Lol, my fan temp in that file was 65000 @lleonid

sterlingrpi, 2020-07-11:
@zorrobyte same issue here. The changing /sys/class/thermal/thermal_zone0/trip_point_4_temp doesn't last after a reboot. Any solution?

aizzaac, 2020-07-31:
@lleonid 

How do you know that: ``` /sys/class/thermal/thermal_zone0/trip_point_4_temp``` is the fan temp?
Where can I find that information?
Also is it possible to monitor: voltage, current or power?

aizzaac, 2020-08-13:
I made a plot. The fan starts working at 65°C. But  do not know if it is the temperature of the CPU or the TPU.

[image]

lleonid, 2020-08-24:
> @lleonid
> 
> How do you know that: ` /sys/class/thermal/thermal_zone0/trip_point_4_temp` is the fan temp?
> Where can I find that information?
> Also is it possible to monitor: voltage, current or power?

It's set in device tree - https://coral.googlesource.com/linux-imx/+/refs/heads/master/arch/arm64/boot/dts/freescale/fsl-imx8mq-som.dtsi#757
And it's bound to CPU temperature.

TPU temperature is available via /sys/class/apex/apex_0/temp

aizzaac, 2020-09-21:
@lleonid 
Can you further explain this:

[image]

```
@brianthelion That will only increase power consumtion by forcing CPU cores to operate at higher frequency.
```

Namburger, 2020-09-21:
@aizzaac cpu scaling doesn't really have much to do with the edgetpu, however it does play a factor in getting the output tensors from the edgetpu (I/O). More on it:
https://wiki.archlinux.org/index.php/CPU_frequency_scaling

ethanyanjiali, 2024-08-02:
Still no solution after 4 years? This should be some simple fix from an overlook somewhere