f0cal/google-coral #16: edgetpu example source code doesn't compile with bazel 0.25.0

TallGuy74, 2019-05-02:
I used [https://coral.googlesource.com/edgetpu](url) to download the edgetpu C++ examples.
This installs bazel 0.25.0 in the docker image (latest version from the bazel debian repo), which introduced a change incompatible with the io_closures used in the workspace.

To work around this, I modified the docker file in tools to install bazel 0.24.1 using the default installer (apt get zip unzip; wget bazel installer, and execute the installer with --prefix=/usr).

Not sure if this is relevant here, if not feel free to close. The edgetpu repo above is referenced in the coral dev board documentation.

TallGuy74, 2019-05-02:
diff --git a/tools/Dockerfile.16.04 b/tools/Dockerfile.16.04
index 511d0a4..ee7c195 100644
--- a/tools/Dockerfile.16.04
+++ b/tools/Dockerfile.16.04
@@ -12,7 +12,7 @@ RUN apt-get ${APT_NONINTERACTIVE} install ca-certificates gnupg
 RUN cat /tmp/bazel_release.pub.gpg | apt-key add -
 RUN echo "deb http://storage.googleapis.com/bazel-apt stable jdk1.8" >> /etc/apt/sources.list
 
-ARG PACKAGES="bazel \
+ARG PACKAGES=" \
     build-essential \
     clang-4.0 \
     crossbuild-essential-arm64 \
@@ -57,5 +57,13 @@ RUN dpkg-deb -R libc++abi-7-dev_7.0.1-8_armhf.deb /tmp
 # Copy .a for statically linkage.
 RUN cp -v /tmp/usr/lib/llvm-7/lib/libc++.a /usr/lib/arm-linux-gnueabihf
 RUN cp -v /tmp/usr/lib/llvm-7/lib/libc++abi.a /usr/lib/arm-linux-gnueabihf
+
+#
+# Workaround for the breaking changes in bazel 0.25.0
+#
+RUN apt install unzip zip
+RUN wget https://github.com/bazelbuild/bazel/releases/download/0.24.1/bazel-0.24.1-installer-linux-x86_64.sh
+RUN bash bazel-0.24.1-installer-linux-x86_64.sh --prefix=/usr
+
 # Clean up.
-RUN rm libc++-7-dev_7.0.1-8_armhf.deb libc++abi-7-dev_7.0.1-8_armhf.deb
\ No newline at end of file
+RUN rm libc++-7-dev_7.0.1-8_armhf.deb libc++abi-7-dev_7.0.1-8_armhf.deb

Namburger, 2020-05-24:
Thanks! Closing for now since we've moved onto new docker images now :)

closed 2020-05-24 by Namburger