f0cal/google-coral #18: Cannot install python libraries in the Coral

caiya55, 2019-05-04:
* Doc you were trying to follow: pip3 install XXX
* Your host OS: ubuntu
* Your Python3 version: python3.6

# What I ran
 pip3 install XXX (tensorflow, scipy, opencv-python, etc.)
# What the docs said should happen
successfully building
# What actually happened
errors, many red errors.

After a few days working, I have successfully run demos on the coral with its camera. It works very well and fast. Now I want to implement my own deep learning code on the coral. I did the following things.

step1. I found the open-pose project is interesting, so I found the pretrained .pd file from https://github.com/ildoonet/tf-pose-estimation. It said the code has already been implemented on Jetxon TX2, so I think it is possible to implement on Coral, maybe with better performance.

step2. I use the tf.contrib.lite.TFliteConverter function to convert the .pb to .tflite, just according to the official guide. And it works well in my Ubuntu system (not working in windows). 

step3. I use  GStreamer samples from the official website as the template to read the camera and process it. 

step4. I revised the template code, so it can estimate pose in real time in my ubuntu system (no GPU). It works.

step5. In this step, I think I could just run the python code in the coral, but I found there are many libraries missing (even the tensorflow !), when I try to install them using pip3 install XXX, I got errors for most libraries. such as opencv-python, scipy, slidingwindow. And the errors are similar, there are many read lines, I am confused about the error but it mentions aarch64-linux-gnu-gcc . 

My question is, since there are many libraries are not installed in the Coral (including tensorflow) what is the correct way the do the implement on the coral. It will be great if someone can give me some clues about the development process. Like how the generate the demos like the official ones, it seems like they don't use python scripts.

brianthelion, 2019-05-04:
@caiya55 We need to triage the specific error modes that the Coral exhibits. Please make a simple `bash` script that demonstrates the issues you're having, and post it here.

caiya55, 2019-05-05:
[image]

caiya55, 2019-05-05:
[image]
[image]

caiya55, 2019-05-05:
I make a screenshot of pip3 install slidingwindow.

brianthelion, 2019-05-05:
`fatal error: Python.h No such file or directory` is a dead giveaway. That means that the C++ headers required for building Python extensions are not installed. You need to `sudo apt-get install python3-dev`.

As a side note, it's potentially problematic to use `sudo` with `pip3`. This practice is generally to be avoided. Do your work in a virtual environment and you don't have to worry about messing up the system-wide Python installation:

```
python3 -m venv _venv # creates the virtual enviorment
source _venv/bin/activate # activates the virtual environment
pip install --upgrade pip
pip install slidingwindow # installs in the virtual environment; no sudo required
...
deactivate # deactivates the virtual environment
```

caiya55, 2019-05-06:
thank you very much. This is perfectly solved my problem. There are several things I'd like to share when I try to solve this problem:
1. I got some weird errors when I try to install libraries. I think reboot the system is a wise choice.
3. after reboot, run sudo apt-get update.
2. For anyone who is from China, it is very necessary to find tools to "cross the wall". For me, shadow socks installed windows, use a command like this: 
sudo apt-get -o Acquire::proxy="your/windows/proxy/address: your shadowsock prots" update

otherwise, the speed will be always zero.

closed 2019-05-06 by brianthelion