f0cal/google-coral #19: `fastboot devices` returns Error

mnguyenngo, 2019-05-06:
* Doc you were trying to follow: https://coral.withgoogle.com/docs/dev-board/get-started/
* Your host OS: Mac OS 10.14.4
* Your Python3 version: Python 3.7.2
* fastboot version: 28.0.2

# What I ran
`fastboot devices`

# What the docs said should happen
`1b0741d6f0609912        fastboot`

# What actually happened
`ERROR: Couldn't create a device interface iterator: (e00002bd)`

Sorry if this is a duplicate. It seems like this issue is being experienced by others on StackExchange (https://android.stackexchange.com/questions/209725/fastboot-devices-command-doesnt-work-after-macos-high-sierra-10-14-4-upgrade/212043). If you can confirm or if you have advice, that would be greatly appreciated.

mnguyenngo, 2019-05-15:
Update on new fastboot version

```
$ fastboot --version
fastboot version 28.0.3-5475833
```

I am unfortunately getting the same error. Is anyone else encountering this issue?

jmidd, 2019-05-24:
Yes same issue here @mnguyenngo did you manage to resolve this?

mnguyenngo, 2019-05-24:
Unfortunately, I have not. I have tried different USB C cables but the results have been the same.

jmidd, 2019-05-24:
I think i might try on a linux image - did you use a USB-A to USB- C cable or a USB-c to USB-c - I'm using a usb-c to usb-c cable which I guess may be the issue

mnguyenngo, 2019-05-24:
I used USB-A to micro-USB and 2 USB-C to USB-Cs. Yeah, it's possible that's the issue.

djg14, 2019-05-28:
I too ran into this problem, but was able to figure a work-around that went like this (not pretty):

1: Download and unzip platform-tools : platform-tools_r26.0.1-darwin.zip
2. Put fastboot into /usr/local/bin
3. Run through DevBoard "Get started" document (https://coral.withgoogle.com/docs/dev-board/get-started/) up to and including the "fastboot 0" command in step of that document. 
4. Open new terminal window as described in "Get Started". Run the "fastboot devices" command. This will fail with fastboot v28, but should work with v26. You will get a return much like that shown in step 5. 
5. Next, rename or  remove fastboot in your /usr/local/bin. 
6. Download and unzip version 28 of platform-tools (platform-tools_r28.0.3-darwin.zip) and put the fastboot file in /usr/local/bin as "fastboot".
7. Go back to your terminal and follow from step 6 on. When I executed the "bash flash.sh" it threw a number of "ERROR: Couldn't create a device interface iterator" errors, but did proceed to the end and my install is working on two Macs--an old MacBookAir (2012) and a new MacMini both running Mojave 10.14.4. The initial workaround described here was done on the Air.

mnguyenngo, 2019-06-24:
Updated to fastboot version 29.0.1-5644136.
I had some initial issues at first, but after switching to a different USB-A to USB-C cable, I did not experience any other issues.

closed 2019-06-24 by mnguyenngo