f0cal/google-coral #21: Environmental sensor board demo not working

bryoneason, 2019-05-09:
* Doc you were trying to follow: [https://coral.withgoogle.com/docs/enviro-board/get-started/]
* Your host OS: <Linux>
* Your Python3 version: <3.5.3>

# What I ran
`sudo python3 enviro_demo.py`
# What the docs said should happen
The doc says you should be able to see sensor data on the OLED and send data to the cloud. But it seems I am unable to detect the crypto processor on the Environmental Sensor Board.

# What actually happened
```
No crypto detected, using SW.
Traceback (most recent call last):
  File "enviro_demo.py", line 76, in <module>
    main()
  File "enviro_demo.py", line 51, in main
    with CloudIot(args.cloud_config) as cloud:
  File "/usr/lib/python3/dist-packages/coral/cloudiot/core.py", line 84, in __init__
    with open(rsa_cert, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: ''
```

brianthelion, 2019-05-09:
@bryoneason This may just be a problem with the doc you were trying to follow. The doc claims
```
[The board] includes a secure cryptoprocessor with Google keys to enable connectivity with Google Cloud IoT Core services, allowing you to securely connect to the device and then collect, process, and analyze the sensor data.
```
Seems like that RSA key is missing, but it doesn't really make sense that Google would bundle a single key inside the `.debs` they had you install with `apt`. They'd want you to have a personalized key. I'm guessing you need to create an account on that [Google IoT Core service](https://cloud.google.com/iot-core/), create the key there, and then install it in your home directory or something.

lleonid, 2019-05-15:
An update has been pushed to resolve a kernel configuration issue.
You need to uninstall previous packages and then update via
sudo apt-get purge python3-coral-enviro
sudo apt autoremove
sudo apt-get update
sudo apt-get upgrade

Then follow https://coral.withgoogle.com/docs/enviro-board/get-started/ again.

JVTEAM, 2019-05-27:
My board is displaying on the OLED, however it isn't giving a value, just the labels.  Also of note, it says "Cloud IoT is disabled per configuration." after the enviro demo is started.

Final note, is there a way to get the data elsewhere besides the OLED?

JVTEAM, 2019-06-04:
configured the .ini file and it seems like the cloud site is seeing it now.  Is there any specific documentation on configuring the cloud portion of the sensor?

KrasowskiMichal, 2019-06-16:
Hi,
I am also going through the demo of ESB setup. I can run the demo code, however, I am not getting the sensor result. Instead on display, I get "nan" value generated by the program (due to "null" value exception).

Environment used:
hardware: Raspberry Pi Zero
```
cat /proc/version
Linux version 4.19.42+ (xx@yy) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1219 Tue May 14 21:16:38 BST 2019
gcc --version
gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
python3 --version
Python 3.5.3
```

Exact commands used:
```
python3 enviro_demo.py
```
(note: using with sudo I have the same error)

Result: (OLED display)
Light: nan lux
Pressure: nan kPa
Temp: nan C
RH nan %
(looped)

Result (stdout):
Cloud IoT is disabled per configuration.
(this is expected, I have not configured IoT yet)

Expected result:
I expected real numbers of temperature, pressure, etc.. to show up at this stage (even without IoT setup).

Suggestions on how to proceed would be welcome ;)

varleyd, 2019-06-19:
> Hi,
> I am also going through the demo of ESB setup. I can run the demo code, however, I am not getting the sensor result. Instead on display, I get "nan" value generated by the program (due to "null" [...]

just to say I've the exact same problem.
Looking at this (board.py) it searches /sys/bus/iio/devices but I've no /sys/bus/iio directory. Should this have been created by the EEPROM on the Environmental Board described in the getting started notes as creating the device tree?

thanks.

ehagerty, 2019-06-20:
Hi, I thought perhaps it was the fact that the docs didn't really make clear (if) it needed I2C and SPI turned on via raspi-config.  However, even now that I've enabled them that way and rebooted, there are still no readings.  Everything says 'nan'.

JVTEAM, 2019-06-25:
This had been previously working on my Coral Dev board.  Now I'm getting a note that says "No crypto detected, using SW." and I'm getting "nan" on my readings.

Bugierek, 2019-06-27:
I also have similar issue. Readings show 'nan' for all the sensors

JVTEAM, 2019-06-27:
FYI, I had the same issue.  Then I tried it on my Coral Dev board and it worked...however days later the nan value returned.  

I contacted support and they told me that "Latest version of Raspbian is breaking few Coral modules and our team is working on to fix that."

They gave me a drive link to a new image.  I tried it on a Pi Zero and it worked again.

https://drive.google.com/open?id=1LiKCdwAbSLWdIQWxqJ5Qoj7jNONKnzMY

Bugierek, 2019-06-27:
Thank you for the info, hopefully they'll resolve it soon.
> They gave me a drive link to a new image. I tried it on a Pi Zero and it worked again.
> 
> https://drive.google.com/open?id=1LiKCdwAbSLWdIQWxqJ5Qoj7jNONKnzMY

Unfortunately the link has a restricted access, is it possible to re-up and pass the link here ?
Thanks

JVTEAM, 2019-06-27:
Its the 2019-04-08-raspbian-stretch.zip release of raspian.  I don't think there is any else special about it.

ehagerty, 2019-06-27:
Hi, I hope the following long description helps someone.

Try as I might, the limited nature of the 'getting started' doc for this board (https://coral.withgoogle.com/docs/enviro-board/get-started/) just wasn't deep or broad enough. However, I've had the great pleasure to be able to work over email with the incredibly helpful support guys at Coral.  Because I have faced all the different problems on this thread, all at different times and in different combinations, I wanted to share my path to success.  After lots of discussion with support, reference to how to docs that covered different bits of what we were trying to achieve, and and hours of me just not being the brightest bulb in the box, we got to the following repeatable set of steps.  

This may seem overkill, but it works for me:

install completely fresh latest raspbian to an sd card
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
sudo reboot
sudo echo "deb https://packages.cloud.google.com/apt coral-cloud-stable main" | sudo tee /etc/apt/sources.list.d/coral-cloud.list
sudo curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
sudo apt install python3-coral-enviro
sudo reboot now

**time to prove to yourself the hat works**
sudo python3 enviro_demo.py

I now had a board working with all four readings, but without being part of the IOT platform, so I edited  /usr/lib/python3/dist-packages/coral/envirocloud_config.ini to match my account details but leaving the encryption empty as suggested in the file.  At this stage I was unable to get enviro_demo.py to run - it crashed out with the error @bryoneason was seeing at the top of this thread where the onboard crypto call in core.py was complaining about a file not found.  

To get around that, I did the following: 

cd /usr/lib/python3/dist-packages/coral/enviro
sudo wget https://pki.goog/roots.pem
sudo openssl genrsa -out rsa_private.pem 2048
sudo openssl rsa -in rsa_private.pem -pubout -out rsa_public.pem
sudo vi cloud_config.ini  <set variable RSACertFile = rsa_public.pem>
sudo vi cloud_config.ini  <set all other variables to match all the rest of your gcloud settings, region etc>

**time to prove to yourself everything works**
sudo python3 enviro_demo.py

For me, I now have a pi, connected to the internet over my local wifi, connected to my gcp account, using the encryption keys I created, which has been running for days, all the way through pubsub and dataflow to bigquery.  It has inserted tens of thousands of rows successfully.

*NB, if you want to do the same thing as a semi-tutorial (accepting i'm hardly a good tutor of anything) you'll need to:

1) enable apis for pubsub, dataflow, bigquery and storage first, then:

2) go to the IOT dashboard, create a new registry (remember the region), create a new device and make sure to add the public key you created above.

3) go to the pubsub dashboard, create a new topic, then create a new subscription (delivery type set to pull) for that topic

4) go back to the topic screen, select your topic, and from the dropdown select 'export to bigquery'

5) now over in the dataflow dashboard, give the flow a name, change the Cloud Dataflow Template from 'topic to bigquery' to 'subscription to bigquery', make sure you set the region to match everything else you are doing, and enter the topic you made above

6) in another tab, go into the bigquery dashboard and create a dataset and a table within that with the following structure:

field name | Type | Mode |
***************************
humidity | FLOAT | NULLABLE 
pressure | FLOAT | NULLABLE 
temperature | FLOAT | NULLABLE 
ambient_light | FLOAT | NULLABLE

7) In another tab go into the Storage Browser, create a bucket and remember its name

8) Now go back to your half complete dataflow, and fill in the output table name with the details you created above and also enter the name of the storage bucket you created and put it in the Temporary Location field and finally set the flow running

9) go back to the bigquery dashboard, and in the query editor, you can use this to start to see your data:
SELECT * FROM `<project id>:<dataset>.<table>` LIMIT 1000

Alone, without the help of these tutorials:

https://cloud.google.com/community/tutorials/cloud-iot-rtdp
https://cloud.google.com/community/tutorials/cloud-iot-gateways-rpi

and the fantastic support of coral-support@google.com plus quite a lot of hours of me assuming I was wrong and hacking at things, this does all work very well indeed.

PS - I've probably missed out some steps or oversimplified others, but hopefully all this is of some assistance to someone.

soungno, 2020-09-12:
There is a bug in the coral-enviro package.

The content of the bug is that the cs_high value setting is missing from ssd1306 (OLED Display) initialization when running on raspberry pi.

Find board.py in the coral enviro package installation directory and modify it as follows.

`class EnviroBoard():
    """
    An interface for all input and output modules on the Environmental Sensor Board.
    """

    def __init__(self):
        # Obtain the full sysfs path of the IIO devices.
        self._hdc2010 = _get_path('hdc20x0')
        self._bmp280 = _get_path('bmp280')
        self._opt3002 = _get_path('opt3001')
        self._tla2021 = _get_path('ads1015')
        # Create SSD1306 OLED instance, with SPI as the interface.
        plat = platform.platform()
        if 'mendel' in plat:
            from .rpi_gpio_periphery import pGPIO
            # Values for the Coral Dev Board (running Mendel Linux).
            # If running legacy kernel (4.9.51), use port 32766.
            port = 32766 if 'Linux-4.9.51-imx' in plat else 0
            self._display = ssd1306(serial_interface=spi(gpio=pGPIO(), port=port, device=0, gpio_DC=138, gpio_RST=140),
                                    gpio=pGPIO(), height=32, rotate=2)
        else:
            # Default to RPi.GPIO in luma and defaults GPIO.
            self._display = ssd1306(serial_interface=spi(**cs_high=True**),
                                    gpio=noop(), height=32, rotate=2)
`

Luisibear98, 2020-11-18:
Hello, I made this work by modifying some parts of the code to directly access the sensors. 
The modify code needs some extra .py classes.
Should I post it somewhere is anybody is interested?

jeffreycunn, 2021-02-06:
soungno, thanks for the insights on the cs_high=True flag in board.py.  This fixed my first error....also main getting started instructions are definitely lacking in detail.  The docs should include enabling SPI and ISC

I have the LED displaying, but now I am getting NAN values....will look above to see if other suggestions fix.

Luisibear98, 2021-02-08:
> soungno, thanks for the insights on the cs_high=True flag in board.py. This fixed my first error....also main getting started instructions are definitely lacking in detail. The docs should include [...]

Some of the libraries are a little bit buggy, thats only one of the bugs