f0cal/google-coral #22: Blink LED Issues

surfnm, 2019-05-20:
Having trouble running a 'Blink LED' program on the Coral Dev Board

Ran updates

  sudo apt-get update
  sudo apt-get dist-upgrade

Loaded python 3 & periphery

  sudo apt-get install python3-pip
  sudo pip3 install python-periphery

Wrote and loaded a Blink LED python program

  #! /usr/bin/python
  from periphery import GPIO
  import time
  gpio = GPIO(138, "out")
  gpio.write(True)
  time.sleep(1)
  gpio.write(False)
  time.sleep(1)
  gpio.write(True)
  time.sleep(1)
  gpio.close()

Attempted to run the python program

  chmod a+x coralBlinkLED.py
  ./coralBlinkLED.py

**Received the following error
  
  ImportError: No module named periphery**

pip3 list shows 

  python-apt (1.4.0b3)
  python-debian (0.1.30)
  python-debianbts (2.6.1)
  python-periphery (1.1.1)

Reran periphery install

  sudo pip3 install python-periphery

**Received the following error

  Requirement already satisfied: python-periphery in /usr/local/lib/python3.5/dist-packages**

Suggestions are appreciated

brianthelion, 2019-05-21:
You're installing things using `pip3`, with the `3` meaning "using `python3`."
```
$ which python3
/usr/bin/python3
$ /usr/bin/python3 --version
Python 3.5.3
```

Meanwhile your script uses `/usr/bin/python`. Note the missing `3`.
```
$ /usr/bin/python --version
Python 2.7.13
```

As a side note, it's potentially problematic to use `sudo` with `pip3`. This practice is generally to be avoided. If you do your work in a virtual environment instead, you don't have to worry about messing up the system-wide Python installation:

```
python3 -m venv _venv # creates the virtual enviorment
source _venv/bin/activate # activates the virtual environment
pip install --upgrade pip
pip install python-periphery # installs in the virtual environment; no sudo required
...
deactivate # deactivates the virtual environment
```

Hammie217, 2019-12-09:
Whilst using periphery within a venv as above i'm getting the following error. Any idea where this is coming from?

Thanks,

```
python3 motorController.py
Traceback (most recent call last):
  File "/home/mendel/.virtualenvs/cv/lib/python3.7/site-packages/periphery/i2c.py", line 135, in transfer
    fcntl.ioctl(self._fd, I2C._I2C_IOC_RDWR, i2c_xfer, False)
OSError: [Errno 6] No such device or address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "motorController.py", line 132, in <module>
    i2c.transfer(0x1d, msgs)
  File "/home/mendel/.virtualenvs/cv/lib/python3.7/site-packages/periphery/i2c.py", line 137, in transfer
    raise I2CError(e.errno, "I2C transfer: " + e.strerror)
periphery.i2c.I2CError: [Errno 6] I2C transfer: No such device or address
```

Namburger, 2020-05-24:
Hi guys - 

I think this was a problem with older version of mendel not allowing users to access these libraries and you'd need to use sudo when calling the libraries. Reflashing the board with the newest image (mendel day 4.0) should fix this issue.

closed 2020-05-24 by Namburger