f0cal/google-coral #23: Error attempting to install driver for Environmental Sensor Board on RPi 2 Model B

smithatlanta, 2019-05-27:
On a Raspberry Pi 2 Model B running Raspbian GNU/Linux 9 (stretch).

Using the install instructions here: https://coral.withgoogle.com/docs/enviro-board/get-started/

I am receiving the following error during the last step of: 

sudo apt install python3-coral-enviro

The error is: 

Error! Bad return status for module build on kernel: 3.6-trunk-rpi (armv7l)
Consult /var/lib/dkms/coral-enviro-drivers/1.3/build/make.log for more information.

The cat of the log file at /var/lib/dkms/coral-enviro-drivers/1.3/build/make.log is:

DKMS make.log for coral-enviro-drivers-1.3 for kernel 3.6-trunk-rpi (armv7l)
Mon May 27 11:45:50 EDT 2019
make: Entering directory '/usr/src/linux-headers-3.6-trunk-rpi'
  LD      /var/lib/dkms/coral-enviro-drivers/1.3/build/built-in.o
  LD      /var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/built-in.o
  LD      /var/lib/dkms/coral-enviro-drivers/1.3/build/light/built-in.o
  LD      /var/lib/dkms/coral-enviro-drivers/1.3/build/adc/built-in.o
  LD      /var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/built-in.o
  CC [M]  /var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.o
  CC [M]  /var/lib/dkms/coral-enviro-drivers/1.3/build/light/opt3001.o
  CC [M]  /var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.o
  CC [M]  /var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.o
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:70:21: error: variable ‘hdc20x0_regmap_config’ has initializer but incomplete type
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:71:2: error: unknown field ‘reg_bits’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:71:2: warning: excess elements in struct initializer [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:71:2: warning: (near initialization for ‘hdc20x0_regmap_config’) [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:72:2: error: unknown field ‘val_bits’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:72:2: warning: excess elements in struct initializer [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:72:2: warning: (near initialization for ‘hdc20x0_regmap_config’) [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:73:2: error: unknown field ‘max_register’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:73:2: warning: excess elements in struct initializer [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:73:2: warning: (near initialization for ‘hdc20x0_regmap_config’) [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:90:3: error: unknown field ‘info_mask_separate’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:94:11: error: ‘IIO_HUMIDITYRELATIVE’ undeclared here (not in a function)
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:96:3: error: unknown field ‘info_mask_separate’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:101:3: error: unknown field ‘info_mask_separate’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c: In function ‘hdc20x0_get_measurement’:
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:122:9: error: implicit declaration of function ‘regmap_get_device’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:122:23: warning: initialization makes pointer from integer without a cast [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c: In function ‘hdc20x0_read_raw’:
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:212:11: error: ‘IIO_VAL_FRACTIONAL’ undeclared (first use in this function)
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:212:11: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:212:4: warning: return makes integer from pointer without a cast [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:213:25: warning: comparison between pointer and integer [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:216:4: warning: return makes integer from pointer without a cast [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c: In function ‘hdc20x0_probe’:
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:268:2: error: implicit declaration of function ‘i2c_check_functionality’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:272:2: error: implicit declaration of function ‘devm_iio_device_alloc’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:272:12: warning: assignment makes pointer from integer without a cast [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:278:2: error: implicit declaration of function ‘devm_regmap_init_i2c’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:278:15: warning: assignment makes pointer from integer without a cast [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:289:2: error: implicit declaration of function ‘devm_iio_device_register’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c: At top level:
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:313:1: warning: data definition has no type or storage class [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:313:1: warning: type defaults to ‘int’ in declaration of ‘module_i2c_driver’ [-Wimplicit-int]
/var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.c:313:1: warning: parameter names (without types) in function declaration [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:92:11: error: ‘IIO_PRESSURE’ undeclared here (not in a function)
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:93:3: error: unknown field ‘info_mask_separate’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:97:3: error: unknown field ‘info_mask_separate’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:129:21: error: variable ‘bmp280_regmap_config’ has initializer but incomplete type
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:130:2: error: unknown field ‘reg_bits’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:130:2: warning: excess elements in struct initializer [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:130:2: warning: (near initialization for ‘bmp280_regmap_config’) [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:131:2: error: unknown field ‘val_bits’ specified in initializer
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:131:2: warning: excess elements in struct initializer [enabled by default]
[... 290 lines cut ...]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:955:12: warning: assignment makes pointer from integer without a cast [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:970:16: error: implicit declaration of function ‘of_device_get_match_data’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1003:2: error: implicit declaration of function ‘devm_regmap_init_i2c’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1003:15: warning: assignment makes pointer from integer without a cast [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1004:2: error: implicit declaration of function ‘IS_ERR’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1006:3: error: implicit declaration of function ‘PTR_ERR’ [-Werror=implicit-function-declaration]
[... cut ...]
/usr/src/linux-headers-3.6-trunk-common/scripts/Makefile.build:443: recipe for target '/var/lib/dkms/coral-enviro-drivers/1.3/build/adc' failed
make[3]: *** [/var/lib/dkms/coral-enviro-drivers/1.3/build/adc] Error 2
/usr/src/linux-headers-3.6-trunk-common/Makefile:1208: recipe for target '_module_/var/lib/dkms/coral-enviro-drivers/1.3/build' failed
make[2]: *** [_module_/var/lib/dkms/coral-enviro-drivers/1.3/build] Error 2
Makefile:130: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
make: Leaving directory '/usr/src/linux-headers-3.6-trunk-rpi'

mbrooksx, 2019-06-04:
Raspbian decided to be difficult and switch from linux-headers to raspberrypi-kernel-headers but leave an antiquated (3.6) linux-headers package available. So if this is installed, modern drivers won't compile (the Enviro drivers compile for all installed kernel versions). Kill that version with fire:

sudo apt purge linux-headers
sudo rm -rf /usr/lib/modules/3.6*

Then you can install.

The package definition has also been changed to only allow installation of linux-headers > 4.9.

smithatlanta, 2019-06-04:
Looks like I'm still having header issues...

When I run the above, I get the following:  

_sudo apt purge linux-headers

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Virtual packages like 'linux-headers' can't be removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded._

I went ahead and ran the _rm -rf /usr/lib/modules/3.6*_
then I did an _apt remove python3-coral-enviro_
and then I did an _apt update_
and reran _apt install python3-coral-enviro_

This was the result:

_sudo apt install python3-coral-enviro

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  coral-enviro-drivers-dkms dkms linux-headers-3.6-trunk-common linux-headers-3.6-trunk-rpi linux-kbuild-3.6 python3-coral-cloudiot python3-cryptoauthlib python3-luma-core python3-luma-oled
  python3-paho-mqtt python3-smbus2
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  coral-enviro-drivers-dkms dkms linux-headers-3.6-trunk-common linux-headers-3.6-trunk-rpi linux-kbuild-3.6 python3-coral-cloudiot python3-coral-enviro python3-cryptoauthlib python3-luma-core
  python3-luma-oled python3-paho-mqtt python3-smbus2
0 upgraded, 12 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/5,001 kB of archives.
After this operation, 30.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
Selecting previously unselected package dkms.
(Reading database ... 133096 files and directories currently installed.)
Preparing to unpack .../00-dkms_2.3-2_all.deb ...
Unpacking dkms (2.3-2) ...
Selecting previously unselected package linux-headers-3.6-trunk-common.
Preparing to unpack .../01-linux-headers-3.6-trunk-common_3.6.9-1~experimental.1+rpi7_armhf.deb ...
Unpacking linux-headers-3.6-trunk-common (3.6.9-1~experimental.1+rpi7) ...
Selecting previously unselected package linux-kbuild-3.6.
Preparing to unpack .../02-linux-kbuild-3.6_3.6-1~experimental.1+rpi2+b2_armhf.deb ...
Unpacking linux-kbuild-3.6 (3.6-1~experimental.1+rpi2+b2) ...
Selecting previously unselected package linux-headers-3.6-trunk-rpi.
Preparing to unpack .../03-linux-headers-3.6-trunk-rpi_3.6.9-1~experimental.1+rpi7_armhf.deb ...
Unpacking linux-headers-3.6-trunk-rpi (3.6.9-1~experimental.1+rpi7) ...
Selecting previously unselected package coral-enviro-drivers-dkms.
Preparing to unpack .../04-coral-enviro-drivers-dkms_1.3_all.deb ...
Unpacking coral-enviro-drivers-dkms (1.3) ...
Selecting previously unselected package python3-paho-mqtt.
Preparing to unpack .../05-python3-paho-mqtt_1.4.0~dev0-1_all.deb ...
Unpacking python3-paho-mqtt (1.4.0~dev0-1) ...
Selecting previously unselected package python3-cryptoauthlib.
Preparing to unpack .../06-python3-cryptoauthlib_20190304-1_armhf.deb ...
Unpacking python3-cryptoauthlib (20190304-1) ...
Selecting previously unselected package python3-coral-cloudiot.
Preparing to unpack .../07-python3-coral-cloudiot_1.0_all.deb ...
Unpacking python3-coral-cloudiot (1.0) ...
Selecting previously unselected package python3-smbus2.
Preparing to unpack .../08-python3-smbus2_0.2.2-1_all.deb ...
Unpacking python3-smbus2 (0.2.2-1) ...
Selecting previously unselected package python3-luma-core.
Preparing to unpack .../09-python3-luma-core_1.8.3-1_all.deb ...
[... 379 lines cut ...]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:970:16: error: implicit declaration of function ‘of_device_get_match_data’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1003:2: error: implicit declaration of function ‘devm_regmap_init_i2c’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1003:15: warning: assignment makes pointer from integer without a cast [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1004:2: error: implicit declaration of function ‘IS_ERR’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1006:3: error: implicit declaration of function ‘PTR_ERR’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1009:2: error: implicit declaration of function ‘devm_iio_triggered_buffer_setup’ [-Werror=implicit-function-declaration]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c: At top level:
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1153:1: warning: data definition has no type or storage class [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1153:1: warning: type defaults to ‘int’ in declaration of ‘module_i2c_driver’ [-Wimplicit-int]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1153:1: warning: parameter names (without types) in function declaration [enabled by default]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:131:36: warning: ‘ads1015_events’ defined but not used [-Wunused-variable]
/var/lib/dkms/coral-enviro-drivers/1.3/build/adc/ti-ads1015.c:1142:26: warning: ‘ads1015_driver’ defined but not used [-Wunused-variable]
cc1: some warnings being treated as errors
[... cut ...]
make[3]: *** [/var/lib/dkms/coral-enviro-drivers/1.3/build/adc] Error 2
/usr/src/linux-headers-3.6-trunk-common/Makefile:1208: recipe for target '_module_/var/lib/dkms/coral-enviro-drivers/1.3/build' failed
make[2]: *** [_module_/var/lib/dkms/coral-enviro-drivers/1.3/build] Error 2
Makefile:130: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
make: Leaving directory '/usr/src/linux-headers-3.6-trunk-rpi'
pi@denpi:/usr/lib/python3/dist-packages/coral/enviro $

mbrooksx, 2019-06-04:
Try removing it again and then explicitly running

sudo apt install raspberrypi-kernel-headers

The Debian install should then see this is installed and skip installing
linux-headers

smithatlanta, 2019-06-04:
Getting much closer.

sudo apt install python3-coral-enviro

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  coral-enviro-drivers-dkms dkms python3-coral-cloudiot python3-cryptoauthlib python3-luma-core python3-luma-oled python3-paho-mqtt python3-smbus2
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  coral-enviro-drivers-dkms dkms python3-coral-cloudiot python3-coral-enviro python3-cryptoauthlib python3-luma-core python3-luma-oled python3-paho-mqtt python3-smbus2
0 upgraded, 9 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/262 kB of archives.
After this operation, 1,463 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Selecting previously unselected package dkms.
(Reading database ... 166466 files and directories currently installed.)
Preparing to unpack .../0-dkms_2.3-2_all.deb ...
Unpacking dkms (2.3-2) ...
Selecting previously unselected package coral-enviro-drivers-dkms.
Preparing to unpack .../1-coral-enviro-drivers-dkms_1.3_all.deb ...
Unpacking coral-enviro-drivers-dkms (1.3) ...
Selecting previously unselected package python3-paho-mqtt.
Preparing to unpack .../2-python3-paho-mqtt_1.4.0~dev0-1_all.deb ...
Unpacking python3-paho-mqtt (1.4.0~dev0-1) ...
Selecting previously unselected package python3-cryptoauthlib.
Preparing to unpack .../3-python3-cryptoauthlib_20190304-1_armhf.deb ...
Unpacking python3-cryptoauthlib (20190304-1) ...
Selecting previously unselected package python3-coral-cloudiot.
Preparing to unpack .../4-python3-coral-cloudiot_1.0_all.deb ...
Unpacking python3-coral-cloudiot (1.0) ...
Selecting previously unselected package python3-smbus2.
Preparing to unpack .../5-python3-smbus2_0.2.2-1_all.deb ...
Unpacking python3-smbus2 (0.2.2-1) ...
Selecting previously unselected package python3-luma-core.
Preparing to unpack .../6-python3-luma-core_1.8.3-1_all.deb ...
Unpacking python3-luma-core (1.8.3-1) ...
Selecting previously unselected package python3-luma-oled.
Preparing to unpack .../7-python3-luma-oled_3.2.0-1_all.deb ...
Unpacking python3-luma-oled (3.2.0-1) ...
Selecting previously unselected package python3-coral-enviro.
Preparing to unpack .../8-python3-coral-enviro_1.0_all.deb ...
Unpacking python3-coral-enviro (1.0) ...
Setting up python3-cryptoauthlib (20190304-1) ...
Setting up python3-paho-mqtt (1.4.0~dev0-1) ...
Setting up python3-smbus2 (0.2.2-1) ...
Setting up dkms (2.3-2) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up python3-luma-core (1.8.3-1) ...
Setting up python3-coral-cloudiot (1.0) ...
Setting up coral-enviro-drivers-dkms (1.3) ...
Loading new coral-enviro-drivers-1.3 DKMS files...
It is likely that 4.19.42-v7+ belongs to a chroot's host
Building for 4.19.42+ and 4.19.42-v7+
Building initial module for 4.19.42+
Error! Bad return status for module build on kernel: 4.19.42+ (armv7l)
Consult /var/lib/dkms/coral-enviro-drivers/1.3/build/make.log for more information.
Setting up python3-luma-oled (3.2.0-1) ...
Setting up python3-coral-enviro (1.0) ...

The make.log

cat /var/lib/dkms/coral-enviro-drivers/1.3/build/make.log
DKMS make.log for coral-enviro-drivers-1.3 for kernel 4.19.42+ (armv7l)
Tue Jun  4 14:11:56 EDT 2019
make: Entering directory '/usr/src/linux-headers-4.19.42+'
  CC [M]  /var/lib/dkms/coral-enviro-drivers/1.3/build/humidity/hdc20x0.o
  CC [M]  /var/lib/dkms/coral-enviro-drivers/1.3/build/light/opt3001.o
[... 30 lines cut ...]
In file included from ./include/linux/linkage.h:7:0,
                 from ./include/linux/kernel.h:7,
                 from ./include/linux/list.h:9,
                 from ./include/linux/module.h:9,
                 from /var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:14:
./include/linux/export.h:16:21: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
 #define THIS_MODULE (&__this_module)
                     ^
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:314:19: note: in expansion of macro ‘THIS_MODULE’
  .driver_module = THIS_MODULE,
                   ^~~~~~~~~~~
./include/linux/export.h:16:21: note: (near initialization for ‘bmp280_info.event_attrs’)
 #define THIS_MODULE (&__this_module)
                     ^
/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.c:314:19: note: in expansion of macro ‘THIS_MODULE’
  .driver_module = THIS_MODULE,
                   ^~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:309: recipe for target '/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.o' failed
make[2]: *** [/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure/bmp280.o] Error 1
scripts/Makefile.build:544: recipe for target '/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure' failed
make[1]: *** [/var/lib/dkms/coral-enviro-drivers/1.3/build/pressure] Error 2
Makefile:1524: recipe for target '_module_/var/lib/dkms/coral-enviro-drivers/1.3/build' failed
make: *** [_module_/var/lib/dkms/coral-enviro-drivers/1.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.19.42+'

rdejana, 2019-06-15:
I have the same issue.  I think it is a kernel driver mismatch as if I stay on the kernel that came with raspbian-2019-04-09, 4.14.98-v7+, I can build and install the kernel mods once I install raspberrypi-kernel-headers.  I'm I try the 4.9 kernel, I get the same error you do.  I haven't tried 4.4.

smithatlanta, 2019-06-16:
I ended up using this script:  https://gist.github.com/psyke83/cb3ca50561480809c246f42727cb7cf2#file-downgrade_firmware-sh
to downgrade my firmware and now I am good to go.  

sudo ~/downgrade_firmware.sh downgrade
sudo ~/downgrade_firmware.sh block
sudo apt remove  python3-coral-enviro
sudo apt-get autoclean
sudo apt-get autoremove
sudo apt install  python3-coral-enviro
sudo reboot

kyletaylored, 2019-06-19:
> I ended up using this script: https://gist.github.com/psyke83/cb3ca50561480809c246f42727cb7cf2#file-downgrade_firmware-sh
> to downgrade my firmware and now I am good to go.
>
> sudo [...]

I've had the same issues reported in this thread, and this worked for me. I folded all this work into a simple script that should help at https://github.com/kyletaylored/coraldev

```bash
curl -fsSL https://raw.githubusercontent.com/kyletaylored/coraldev/master/install.sh | bash
```

Namburger, 2020-05-24:
Super late here, but the kernal-headers is a dependency for this package. This is maintained by the vendors (rpi for this example) so we won't really be able to fix this one.

closed 2020-05-24 by Namburger