f0cal/google-coral #24: Error when running Inception v3 or v4 on coral dev board

1n1n1t3, 2019-05-28:
* Doc you were trying to follow: the official one, I just added the Inception v4 edgetpu model from https://coral.withgoogle.com/models/
* Your host OS: stock last version 
* Your Python3 version: stock/unchanged

# What I ran
python3 classify_server.py --model inception_v4_299_quant_edgetpu.tflite --labels imagenet_labels.txt

# What the docs said should happen
Run a server streaming with classification from coral camera feed on port 4664

# What actually happened
The streaming server starts, but when I attempt to open it in browser e.g. 192.168.0.104:4664 I get this error in the shell:
ERROR:gstglbasememory.c:94:_mem_create_gl: assertion failed: (error != NULL)
Aborted

I get the error only for inception v3 and v4. v1 and v2 are working. Any idea ? I guess it's something about the memory on the device but then again I see that there were benchmarks run on the device (the dev board) with these models.

JVTEAM, 2019-06-21:
I get the same error.  However, I run the command edgetpu_classify_server with the model and labels tag rather than the python command you are using.  Where did you get the classify_server.py script?  Want to try it on the usb accelerator.

Ferdostassa, 2019-09-09:
Hi, i ave got exact same error. I get a Cam black screen , when I attempt to open it in browser e.g. 192.168.100.2:4664  with Incetionv3.

Appreciate if anyone can shed light as how to resolve the issue.

Namburger, 2020-05-24:
Inception are segmentation models, they are not compatible with classification model. Use this example for segmentation: https://github.com/google-coral/edgetpu/blob/master/examples/semantic_segmetation.py

closed 2020-05-24 by Namburger