f0cal/google-coral #25: install.sh says platform is not supported, but it is

MartinSmeyer, 2019-05-30:
I have this linux system (Ubuntu 16.04):  
```
uname -a  
```

Linux martin-Lenovo-YOGA-910-13IKB 4.9.0-040900-generic #201612111631 SMP Sun Dec 11 21:33:00 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux  

```
OS_VERSION=$(uname -v)
```
#201612111631 SMP Sun Dec 11 21:33:00 UTC 2016

this does not contain "Ubuntu" or "Debian" as asked for in l46 in install.sh

Removing 
```
&& [[ "${OS_VERSION}" == *"Debian"* || "${OS_VERSION}" == *"Ubuntu"* ]]
```
solves the problem.

brianthelion, 2019-05-30:
Can you provide some more context? What is `install.sh`?

Namburger, 2020-05-24:
@MartinSmeyer This shouldn't be a problem anymore as we have now packaged our libraries as apt packages :)

closed 2020-05-24 by Namburger