f0cal/google-coral #27: MDT not compatible with ARM64

JVTEAM, 2019-06-11:
* Doc you were trying to follow: https://coral.withgoogle.com/docs/dev-board/mdt/
* Your host OS: ubuntu 18.04 (on jetson nano)
* Your Python3 version: 3.6

# What I ran

# Add the repo:
echo "deb https://packages.cloud.google.com/apt coral-mdt-stable main" | sudo tee /etc/apt/sources.list.d/mdt.list

# Add Google keys
curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -

# Then install the package:
sudo apt-get update

sudo apt-get install mendel-development-tool

# What the docs said should happen
MDT installed

# What actually happened
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://packages.cloud.google.com/apt coral-mdt-stable InRelease' doesn't support architecture 'arm64'

I was trying to make it so I could control the board with my Jetson Nano....apparently its not compatible with the same architecture that the coral board is....

lleonid, 2019-10-08:
pip3 install --user mendel-development-tool
is the way on arm64 platforms.

JVTEAM, 2019-10-17:
Doesn't work on the Jetson Nano

Namburger, 2020-05-24:
@JVTEAM hummn, what error messages do you get? This is mostlikely just because the pip package wasn't packaged for arm platform. Try downloading it from here:
https://pypi.org/project/mendel-development-tool/

Furthermore, mdt is a user friendly wrapper around ssh, you can also use ssh to access the board!

closed 2020-05-24 by Namburger