f0cal/google-coral #31: installation fails on new Buster release - dependency problem

KrasowskiMichal, 2019-06-24:
* Doc you were trying to follow: https://coral.withgoogle.com/docs/enviro-board/get-started/
* Your host OS: 
```
cat /proc/version
Linux version 4.19.50+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #896 Thu Jun 20 16:09:52 BST 2019
```
Raspbian Buster with desktop
Image with desktop based on Debian Buster
Version:June 2019
Release date:2019-06-20
Kernel version:4.19

* Your Python3 version: 
```
python3 --version
Python 3.7.3
```

# What I ran
```
echo "deb https://packages.cloud.google.com/apt coral-cloud-stable main" | sudo tee /etc/apt/sources.list.d/coral-cloud.list
curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
sudo apt update
sudo apt upgrade
pi@pizerow:~ $ sudo apt install python3-coral-enviro
```

# What the docs said should happen
(package installed ;)

# What actually happened
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-coral-enviro : Depends: python3-coral-cloudiot but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

This seems unreported yet, caused by OS new release.

Namburger, 2020-05-24:
Ahh this should no longer be a problem as the package is usable for Mendel Day(4.0) which is a derivative of buster + I tested this on my Pi. Closeing for now, please update me if it's still broken for you

closed 2020-05-24 by Namburger

ries9112, 2020-05-31:
@Namburger I just ran into this problem exactly as described by @KrasowskiMichal. Thoughts on how I would go about fixing this?

Namburger, 2020-06-01:
@ries9112 what's your`uname -a`