f0cal/google-coral #32: Installing Docker on Google Coral Dev Board

adr-arroyo, 2019-06-25:
Hello,

I would like to install docker in the Google Coral Dev board. Official support says that they have not tried to do it, but "I can try to do it". 

Anyway, there is no "official" or supported way to install docker in the board, mainly because the get.docker.com script does not work on the coral.

Has anyone tried to install it by source? Following the [official steps](https://docs.docker.com/install/linux/docker-ce/binaries/). The first problem is the `iptables` command, it is not in the coral SO.

Cheers

adr-arroyo, 2019-06-25:
Update:

This might sound weird. Even though we cannot get docker from the website. It is completely possible to run docker from the Coral Dev Board. 

Just follow the steps in the official tutorial for getting Docker CE from binaries [here](https://docs.docker.com/install/linux/docker-ce/binaries/). You need to run:

```
$ wget https://download.docker.com/linux/static/stable/armhf/docker-18.09.6.tgz 
``` 

To get the compatible arm version of docker. If you follow the rest of the tutorial, you should have docker running on your dev board. 

It is highly probable that you need to run `sudo dockerd &` (docker daemon) every time you restart the coral. In addition, you might want to follow [post-installation steps](https://docs.docker.com/install/linux/linux-postinstall/) for running docker without sudo.

Cheers!

mmortazavi, 2019-11-13:
I am struggling to make up and running. From @adr-arroyo recommendation I managed to install the docker successfully on the board, and hello-world works too. In my case I am trying to run the `gcr.io/automl-vision-ondevice/gcloud-container-1.12.0` (see [here](https://cloud.google.com/vision/automl/docs/containers-gcs-tutorial)) on the board for my custom object detection model. Everything works fine on my local machine, but `docker run` based on the above mentioned link like:

`sudo docker run --rm --name ${CONTAINER_NAME} -p ${PORT}:8501 -v ${YOUR_MODEL_PATH}:/tmp/mounted_model/0001 -t ${CPU_DOCKER_GCS_PATH}`

Keeps returning strange errors (in DEBU mode):

```
errored_model/0001 -t gcr.io/automl-vision-ondevice/gcloud-container-1.12.0
DEBU[2019-11-13T16:56:26.798785994Z] Calling GET /_ping
DEBU[2019-11-13T16:56:26.802488484Z] Calling POST /v1.39/containers/create?name=tf-serving_datamatrix
DEBU[2019-11-13T16:56:26.803478967Z] form data: {"AttachStderr":true,"AttachStdin":false,"AttachStdout":true,"Cmd":null,"Domainname":"","Entrypoint":null,"Env":[],"ExposedPorts":{"8501/tcp":{}},"HostConfig":{"AutoRemove":true,"Binds":["/home/mendel/datamatrix-serving-gcp/asset:/tmp/mounted_model/0001"],"BlkioDeviceReadBps":null,"BlkioDeviceReadIOps":null,"BlkioDeviceWriteBps":null,"BlkioDeviceWriteIOps":null,"BlkioWeight":0,"BlkioWeightDevice":[],"CapAdd":null,"CapDrop":null,"Cgroup":"","CgroupParent":"","ConsoleSize":[0,0],"ContainerIDFile":"","CpuCount":0,"CpuPercent":0,"CpuPeriod":0,"CpuQuota":0,"CpuRealtimePeriod":0,"CpuRealtimeRuntime":0,"CpuShares":0,"CpusetCpus":"","CpusetMems":"","DeviceCgroupRules":null,"Devices":[],"DiskQuota":0,"Dns":[],"DnsOptions":[],"DnsSearch":[],"ExtraHosts":null,"GroupAdd":null,"IOMaximumBandwidth":0,"IOMaximumIOps":0,"IpcMode":"","Isolation":"","KernelMemory":0,"Links":null,"LogConfig":{"Config":{},"Type":""},"MaskedPaths":null,"Memory":0,"MemoryReservation":0,"MemorySwap":0,"MemorySwappiness":-1,"NanoCpus":0,"Networ[ 2686.209220] docker0: port 1(veth4740dd6) entered blocking state
kMode":"default","OomKillDisable"[ 2686.217511] docker0: port 1(veth4740dd6) entered disabled state
:false,"OomScoreAdj":0,"PidMode":[ 2686.227186] device veth4740dd6 entered promiscuous mode
"","PidsLimit":0,"PortBindings":{"8501/tcp":[{"HostIp":"","HostPo[ 2686.239157] IPv6: ADDRCONF(NETDEV_UP): veth4740dd6: link is not ready
rt":"8501"}]},"Privileged":false,"PublishAllPorts":false,"ReadonlyPaths":null,"ReadonlyRootfs":false,"RestartPolicy":{"MaximumRetryCount":0,"Name":"no"},"SecurityOpt":null,"ShmSize":0,"UTSMode":"","Ulimits":null,"UsernsMode":"","VolumeDriver":"","VolumesFrom":null},"Hostname":"","Image":"gcr.io/automl-vision-ondevice/gcloud-container-1.12.0","Labels":{},"NetworkingConfig":{"EndpointsConfig":{}},"OnBuild":null,"OpenStdin":false,"StdinOnce":false,"Tty":true,"User":"","Volumes":{},"WorkingDir":""}
DEBU[2019-11-13T16:56:26.849447978Z] container mounted via layerStore: &{/var/lib/docker/overlay2/404baa115bb65d34ef72de7ff8cb457f226dcb1b9c4aa091b5d8c338e7880943/merged 0xaaaab56b1f00 0xaaaab56b1f00}
DEBU[2019-11-13T16:56:26.889159331Z] Calling POST /v1.39/containers/ceb1b2220684126fc29047f71eff19e82a6005ce97f493baee1ae5b9bd3b6584/attach?stderr=1&stdout=1&stream=1
DEBU[2019-11-13T16:56:26.889558692Z] attach: stdout: begin
DEBU[2019-11-13T16:56:26.889629732Z] attach: stderr: begin
DEBU[2019-11-13T16:56:26.891183977Z] Calling POST /v1.39/containers/ceb1b2220684126fc29047f71eff19e82a6005ce97f493baee1ae5b9bd3b6584/wait?condition=removed
DEBU[2019-11-13T16:56:26.893797824Z] Calling POST /v1.39/containers/ceb1b2220684126fc29047f71eff19e82a6005ce97f493baee1ae5b9bd3b6584/start
DEBU[2019-11-13T16:56:26.900843164Z] container mounted via layerStore: &{/var/lib/docker/overlay2/404baa115bb65d34ef72de7ff8cb457f226dcb1b9c4aa091b5d8c338e7880943/merged 0xaaaab56b1f00 0xaaaab56b1f00}
DEBU[2019-11-13T16:56:26.901884767Z] Assigning addresses for endpoint tf-serving_datamatrix's interface on network bridge
DEBU[2019-11-13T16:56:26.901957847Z] RequestAddress(LocalDefault/172.17.0.0/16, <nil>, map[])
DEBU[2019-11-13T16:56:26.902039927Z] Request address PoolID:172.17.0.0/16 App: ipam/default/data, ID: LocalDefault/172.17.0.0/16, DBIndex: 0x0, Bits: 65536, Unselected: 65533, Sequence: (0xc0000000, 1)->(0x0, 2046)->(0x1, 1)->end Curr:3 Serial:false PrefAddress:<nil>
DEBU[2019-11-13T16:56:26.957860006Z] Assigning addresses for endpoint tf-serving_datamatrix's interface on network bridge
DEBU[2019-11-13T16:56:26.970535882Z] Programming external connectivity on endpoint tf-serving_datamatrix (67a977d6e6912139fb5cafb74c3ca18abc1c63b8153e077982568c0fe4f4cd0d)
DEBU[2019-11-13T16:56:26.970758123Z] /sbin/iptables, [--wait -t nat -C DOCKER -p tcp -d 0/0 --dport 8501 -j DNAT --to-destination 172.17.0.2:8501 ! -i docker0]
DEBU[2019-11-13T16:56:26.983726080Z] /sbin/iptables, [--wait -t nat -A DOCKER -p tcp -d 0/0 --dport 8501 -j DNAT --to-destination 172.17.0.2:8501 ! -i docker0]
DEBU[2019-11-13T16:56:26.997187478Z] /sbin/iptables, [--wait -t filter -C DOCKER ! -i docker0 -o docker0 -p tcp -d 172.17.0.2 --dport 8501 -j ACCEPT]
DEBU[2019-11-13T16:56:27.012740602Z] /sbin/iptables, [--wait -t filter -A DOCKER ! -i docker0 -o docker0 -p tcp -d 172.17.0.2 --dport 8501 -j ACCEPT]
DEBU[2019-11-13T16:56:27.026378641Z] /sbin/iptables, [--wait -t nat -C POSTROUTING -p tcp -s 172.17.0.2 -d 172.17.0.2 --dport 8501 -j MASQUERADE]
[... cut ...]
`Linux jumping-snail 4.9.51-imx #1 SMP PREEMPT Tue May 14 20:34:37 UTC 2019 aarch64 GNU/Linux
`

And tried to install docker compatible with the aarch64 architecture from directly [Docker static distribution ](https://download.docker.com/linux/static/stable/aarch64/), but still throws the same annoying error. 

Have anyone, @adr-arroyo maybe you, managed to have a docker container running on the board (besides hello-world)? I am wondering if I have other nasty permisison issues that I do not see!

adr-arroyo, 2019-11-14:
Hello @mmortazavi 

I am sorry but I do not remember if I made it work any other container on the coral. I had to stop working on it. However, I do not remember seen this errors, did you follow the steps for the arm installation?

Regards

mmortazavi, 2019-11-14:
Hi @adr-arroyo Thanks for your quick feedback. I see. I thought you continued working on it. Yes I did follow the arm installation. I have already tried a few ways to run docker images customized for my use case, none of them worked! Would you mind opening this ticket again and perhaps change the title to 'Install and Run on Google Coral Dev Board'? I rather have this in this issue and not open another issue. Maybe others see and offer help. Regards

adr-arroyo, 2019-11-15:
It's done @mmortazavi :)

Hopefully there is someone else trying this and has some more info for us.

mmortazavi, 2019-11-15:
Thanks @adr-arroyo. For some reasons I thought this repo is the official google-coral one, but it is not! I may bring the question to the original repo!

adr-arroyo, 2019-11-15:
It's okay @mmortazavi  ;)

But please, reference this issue on the comments of the issue in the original repo!

Namburger, 2019-12-12:
Hi guys, Nam from Coral team here, installing docker binary is pretty straight forward for mendel day:
```
$ wget https://download.docker.com/linux/static/stable/aarch64/docker-19.03.5.tgz (optionally go [here](https://download.docker.com/linux/static/stable/aarch64/docker-19.03.5.tgz) for more packages)
$ tar xvf docker-19.03.5.tgz
$ sudo cp docker* /usr/bin/
```
That's all.

if you want to run docker, first make sure the docker daemon is up with (you'll need to do this every time you reboot the board):
```
$ sudo dockerd & 
```
test:
```
$ sudo docker run hello-world
```
Do this if you don't have the desire to sudo anything:
https://docs.docker.com/install/linux/linux-postinstall/

Namburger, 2019-12-13:
Here is also an example Dockerfile:
https://gist.github.com/Namburger/ceb384626657963f27da18d8a5080486

ramunelluri184, 2019-12-18:
> Hi guys, Nam from Coral team here, installing docker binary is pretty straight forward for mendel day:
>
> ```
> $ wget https://download.docker.com/linux/static/stable/aarch64/docker-19.03.5.tgz [...]

Hey.. I am trying to configure kubernetes on coral board, which depends on docker installation. I followed your comment and installed docker. everything works but i could not start the "_docker.service_" through "_systemctl_" command, in turn kubernetes throws the same error. could you please help me out in this case?

Namburger, 2020-01-07:
@ramunelluri184 my apologies, I didn't see this earlier. I'm not too familiar with kubernetes, but if you think the docker binary installation method from above is causing this issue, you can try this instead (this way you can have the full debian package):
```
$ sudo apt-get install apt-transport-https ca-certificates curl gnupg2 software-properties-common
$ curl -fsSL https://download.docker.com/linux/debian/gpg | sudo apt-key add -
$ echo "deb [arch=arm64] https://download.docker.com/linux/debian buster stable" | sudo tee /etc/apt/sources.list.d/docker.list
$ sudo apt install docker-ce docker-ce-cli containerd.io
```
Hope this helps!

ramunelluri184, 2020-01-14:
> @ramunelluri184 my apologies, I didn't see this earlier. I'm not too familiar with kubernetes, but if you think the docker binary installation method from above is causing this issue, you can try this instead (this way you can have the full debian package):
> 
> ```
> $ sudo apt-get install apt-transport-https ca-certificates curl gnupg2 software-properties-common
> $ curl -fsSL https://download.docker.com/linux/debian/gpg | sudo apt-key add -
> $ echo "deb [arch=arm64] https://download.docker.com/linux/debian buster stable" | sudo tee /etc/apt/sources.list.d/docker.list
> $ sudo apt update
> $ sudo apt install docker-ce docker-ce-cli containerd.io
> ```
> 
> Hope this helps!

Thanks a lot. this worked like a charm and docker is as _systemd_ service.

closed 2020-05-24 by Namburger

saket424, 2021-01-18:
@Namburger 
I got a new google coral dev miniboard and am having trouble installing docker on  it. I see an error

ERRO[2021-01-18T19:49:27.439668055Z] 'overlay' not found as a supported filesystem on this host. Please ensure kernel is new enough and has overlay support loaded.  storage-driver=overlay

Any ideas ?

saket424, 2021-01-19:
I suspect the right docker related flags for kernel support have not been enabled for this new mediatek coral dev miniboard yet. the regular dev board works fine