f0cal/google-coral #35: Issue when installing MDT

libo-wu, 2019-07-13:
* Doc you were trying to follow: https://coral.withgoogle.com/docs/dev-board/get-started/
* Your host OS: Linux
* Your Python3 version: 3.5.3

# What I ran
I followed the getting started document step by step, until I ran **install the Mendel Development Tool (MDT)** with
`pip3 install --user mendel-development-tool`

# What actually happened
`Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-x6125sm9/pynacl/setup.py", line 255, in <module>
        "Programming Language :: Python :: 3.7",
      File "/home/mendel/.local/lib/python3.5/site-packages/setuptools/__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "/home/mendel/.local/lib/python3.5/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/home/mendel/.local/lib/python3.5/site-packages/setuptools/dist.py", line 717, in fetch_build_eggs
        replace_conflicting=True,
      File "/home/mendel/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 782, in resolve
        replace_conflicting=replace_conflicting
      File "/home/mendel/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 1065, in best_match
        return self.obtain(req, installer)
      File "/home/mendel/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 1077, in obtain
        return installer(requirement)
      File "/home/mendel/.local/lib/python3.5/site-packages/setuptools/dist.py", line 784, in fetch_build_egg
        return cmd.easy_install(req)
      File "/home/mendel/.local/lib/python3.5/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/home/mendel/.local/lib/python3.5/site-packages/setuptools/command/easy_install.py", line 705, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/home/mendel/.local/lib/python3.5/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/home/mendel/.local/lib/python3.5/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/home/mendel/.local/lib/python3.5/site-packages/setuptools/command/easy_install.py", line 1146, in run_setup
        raise DistutilsError("Setup script exited with %s" % (v.args[0],))
    distutils.errors.DistutilsError: Setup script exited with error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-x6125sm9/pynacl/`

# Then I update the setup-tools
`pip3 install --upgrade setuptools`
But got the same error message.

JVTEAM, 2019-07-14:
What computer are you install MDT on?

libo-wu, 2019-07-14:
> What computer are you install MDT on?

I use a virtual machine with Ubuntu OS to establish the serial with the TPU dev board.

rahul4tripathi2, 2019-10-16:
Please run this command and see it worked or not
sudo apt-get install python3 python-dev python3-dev \
     build-essential libssl-dev libffi-dev \
     libxml2-dev libxslt1-dev zlib1g-dev \
     python-pip

closed 2020-05-24 by Namburger