f0cal/google-coral #37: TPU unusable- provided shared library refernences undefined symbols: _Unwind_SetIP

jleibund, 2019-08-07:
* Doc you were trying to follow: undefined symbol: _Unwind_SetIP
* Your host OS: Raspbian GNU/Linux 9 (stretch)
* Your Python3 version: Python 3.5.3

# What I ran
./install.sh
... <omitting throttling message> ...
Reading package lists... Done
Building dependency tree        
Reading state information... Done
libc++1 is already the newest version (3.5-2).
libc++abi1 is already the newest version (3.5-2).
libgcc1 is already the newest version (1:6.3.0-18+rpi1+deb9u1).
libunwind8 is already the newest version (1.1-4.1).
libusb-1.0-0 is already the newest version (2:1.0.21-1).
python3-numpy is already the newest version (1:1.12.1-3).
python3-pil is already the newest version (4.0.0-4).
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Done.
Installing device rule file [/etc/udev/rules.d/99-edgetpu-accelerator.rules]...
File already exists. Replacing it...
Done.
Installing Edge TPU runtime library [/usr/lib/arm-linux-gnueabihf/libedgetpu.so.1.0]...
File already exists. Replacing it...
Done.
Installing Edge TPU Python API...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: edgetpu==2.11.1 from file:///home/pi/edgetpu_api/edgetpu-2.11.1-py3-none-any.whl in /usr/local/lib/python3.5/dist-packages (2.11.1)

(openvino) pi@navio:~/edgetpu_api $ python
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from tflite_runtime.interpreter import load_delegate
>>> load_delegate('libedgetpu.so.1.0')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.virtualenvs/openvino/lib/python3.5/site-packages/tflite_runtime/interpreter.py", line 165, in load_delegate
    delegate = Delegate(library, options)
  File "/home/pi/.virtualenvs/openvino/lib/python3.5/site-packages/tflite_runtime/interpreter.py", line 89, in __init__
    self._library = ctypes.pydll.LoadLibrary(library)
  File "/usr/lib/python3.5/ctypes/__init__.py", line 425, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib/python3.5/ctypes/__init__.py", line 347, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /usr/lib/arm-linux-gnueabihf/libc++abi.so.1: undefined symbol: _Unwind_SetIP

# What the docs said should happen
libedgetpu.so.1.0 should have loaded to be used when creating the Interpreter, e.g.

interpreter = Interpreter(model_path=model_path,experimental_delegates=[load_delegate('libedgetpu.so.1.0')])

# What actually happened
The provided shared library references undefined symbols, presumably, that were expected in these versions of libunwind8 and libc++abi1.  See below.

>  ldd -r /usr/lib/arm-linux-gnueabihf/libc++abi.so.1
	linux-vdso.so.1 (0x7eff5000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76e78000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76e38000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76cf9000)
	/lib/ld-linux-armhf.so.3 (0x76ee7000)
undefined symbol: _Unwind_GetRegionStart	(/usr/lib/arm-linux-gnueabihf/libc++abi.so.1)
undefined symbol: _Unwind_RaiseException	(/usr/lib/arm-linux-gnueabihf/libc++abi.so.1)
undefined symbol: _Unwind_SetIP	(/usr/lib/arm-linux-gnueabihf/libc++abi.so.1)
undefined symbol: _Unwind_GetLanguageSpecificData	(/usr/lib/arm-linux-gnueabihf/libc++abi.so.1)
undefined symbol: _Unwind_GetIP	(/usr/lib/arm-linux-gnueabihf/libc++abi.so.1)
undefined symbol: _Unwind_SetGR	(/usr/lib/arm-linux-gnueabihf/libc++abi.so.1)
undefined symbol: _Unwind_DeleteException	(/usr/lib/arm-linux-gnueabihf/libc++abi.so.1)

nopphons, 2019-12-17:
@jleibund any updates on this? I got the same issue. Have you solved it already?

Namburger, 2020-05-24:
Apologies for this!
This issue should be fixed now as we've started shipping `libedgetpu.so` as a debian package!
```
$ sudo apt install libedgetpu1-std
```

closed 2020-05-24 by Namburger