f0cal/google-coral #38: Trying to install Coral Enviro Board to a Pi Zero W

kflmiami420, 2019-08-09:
Purchased from Mouser.com   08-05-2019

* Doc you were trying to follow: The Instructions printed in the package state " To get started, follow the instructions at   g.co/coral/setup  it does nothing !!

* Your host OS: Debian Strech April release updated and upgraded 
* Your Python3 version: python 3.4 

# What I ran
from https://coral.withgoogle.com/docs/enviro-board/get-started/

python3 enviro_demo.py

# What actually happened

Screen shows Temp : nan C
                       RH     : nan %
                       Light  : Lux            
                   Pressure: kPa

no numbers

maartenterpstra, 2019-08-10:
I can confirm this issue. I get the same readings with when having followed the same steps. I should also note that it also does not work on Raspbian Buster and SPI and I2C are enabled in `raspi-config`. 

My only difference is that even `i2cdetect` fails to detect anything.

kflmiami420, 2019-08-12:
are you running    "  i2cdetect -y 1 "

kflmiami420, 2019-08-18:
trying again a new fresh install 

I get this  pi@raspberrypize10:~ $ echo "deb https://packages.cloud.google.com/apt coral-cloud-stable main" | sudo tee /etc/apt/sources.list.d/coral-cloud.list
deb https://packages.cloud.google.com/apt coral-cloud-stable main
pi@raspberrypize10:~ $ ls
Desktop    Downloads  Music      Pictures  Templates
Documents  MagPi      mypi-2.py  Public    Videos
pi@raspberrypize10:~ $ echo "deb https://packages.cloud.google.com/apt coral-cloud-stable main" | sudo tee /etc/apt/sources.list.d/coral-cloud.list
deb https://packages.cloud.google.com/apt coral-cloud-stable main
pi@raspberrypize10:~ $ curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   659  100   659    0     0    569      0  0:00:01  0:00:01 --:--:--   569
OK
pi@raspberrypize10:~ $ sudo apt install python3-coral-enviro
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python3-coral-enviro

kflmiami420, 2019-08-20:
Update I was emailed by support from Coral Enviro sensor board and they suggested I run sudo apt update/upgrade .   I did and that Unable to locate package python3-coral-enviro error went away and not I get a new error.   But the board is not functioning

kflmiami420, 2019-08-22:
Update I finally after re formatting multiple times  update and upgrade finally it works .   

I would like now to change a few things like convert C to F   and kPa to mbar 

does anyone know the file to make those changes in ?

maartenterpstra, 2019-08-25:
I also got mine working. Turns out the Pi Zero W GPIO interface was broken. FYI, I got it working by installing Raspbian Stretch, running a `sudo apt update && sudo apt upgrade`, following the getting started guide and before running the demo enabling the SPI and I2C interface using `sudo raspi-config`. 

@kflmiami420 
Regarding your last question, in the `enviro_demo.py` file they show how to read the temperature and the pressure. They always read in celcius and kPa according to the datasheet.  For celcius to fahrenheit you can do `enviro.temperature * (9.0 / 5.0) + 32` and for kPa to mbar you can do `enviro.pressure * 10.0`. And change the units in your `msg` statements, of course.

kflmiami420, 2019-08-26:
Thank you ,   I am looking to add some sensors via the I2C plug , but I have a question when I run from commad   i2cdetect -y 1   I get a lot of UU   instead of the proper address for the sensor .  Do you get the same readings on your board ?

kflmiami420, 2019-08-26:
this is what I get now 

pi@raspberrypize10:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: UU -- -- -- -- UU -- -- -- UU -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- UU --                         
pi@raspberrypize10:~ $

kflmiami420, 2019-08-26:
I wanted to share this link to my post on another site forum about this board 

https://www.raspberrypi.org/forums/viewtopic.php?f=45&t=249302

Namburger, 2020-05-24:
@kflmiami420 did you enable `i2c` as well as `gpio` using the `rspi-config` advance setting?
You can learn more about that here: https://stackoverflow.com/questions/17821063/why-does-the-i2cdetect-always-gives-uu-on-my-rtc-in-embedded-linux
Any how, i2c giving UU is an indication that the pi can't access the i2c interface, which is required to interact with the enviro sensor!

closed 2020-05-24 by Namburger