f0cal/google-coral #40: Can't install updates on Dev Board

scch-sims, 2019-08-12:
* Doc you were trying to follow: [Failed_to_Update.txt](https://github.com/f0cal/google-coral/files/3493191/Failed_to_Update.txt)
* Your host OS: Mendel OS (3.0 Chef)

Hello!

I have got a problem with updating my Google Coral Dev Board.
With the command:
sudo apt-get update
it won't work. => have a look at my attached file.
Because of this it is not working to install the edgetpu compiler. :/

EspenTeigen, 2019-09-09:
Same problem here. Any solution?

scch-sims, 2019-09-09:
It worked for me after I used an unlimited network over the hotspot of my phone ^^
I just had the problem that the companies network was restricted.

Hope you can solve it!

closed 2019-09-09 by scch-sims

EspenTeigen, 2019-09-09:
I was on mobilenetwork for the same reason. Ended up re-flashing the dwv board, it works now.