f0cal/google-coral #42: ImportError: No module named 'edgetpu'

zubairahmed-ai, 2019-08-20:
* Doc you were trying to follow: https://coral.withgoogle.com/docs/accelerator/get-started/ 
https://www.pyimagesearch.com/2019/04/22/getting-started-with-google-corals-tpu-usb-accelerator/

* Your host OS: Raspbian on RPi 3B+
* Your Python3 version: 3.5.2

# What I ran
python3 demo/classify_image.py  --model ~/edgetpu_models/mobilenet_v2_1.0_224_quant_edgetpu.tflite  --label ~/edgetpu_models/imagenet_labels.txt --image test_data/parrot.jpg 
# What the docs said should happen
I should see the results of the parrot classification
macaw
Score :  0.99609375

# What actually happened

File "demo/classify_image.py", line 3, in <module> 
from edgetpu.classification.engine import ClassificationEngine
**ImportError: No module named 'edgetpu'**

zubairahmed-ai, 2019-08-20:
This is due to the __init__.py not doing its job

As a workaround I had to remove `edgetpu` from all the import statements

`import edgetpu.classification.engine`

becomes

`import classification.engine`

and so on, anyone knows how to fix this?

JeanCarm, 2019-10-26:
Hi. Did you figure out the problem? I'm having the exat problem. Can you help me please?

PjotrG, 2019-11-02:
Getting exact the same comment, any idea????

JeanCarm, 2019-11-02:
Hey man, I reinstalled the edgetpu library and it worked for me. https://coral.withgoogle.com/docs/edgetpu/api-intro/
Let me know if it works and also what type of project are you working on?
@PjotrG

PjotrG, 2019-11-02:
Thanks Jean for quick response, I tried already to reinstall the

library -but will try again tomorrow..

Concerning projects: I just started with the Coral Edge TPU and installed

Mike Tykas teachable machine --which worked fine in the beginning

but failed now (very frustraiting).

I planned to combine the TPU with a robot arm or the Monsterborg

but just started....

I will try again and will let you know

Thanks again

Peter

 

 

Gesendet: Samstag, 02. November 2019 um 18:21 Uhr

PjotrG, 2019-11-03:
Jean, thanx once again for your advice.

Tried once again to upload, got this:

 

pi@raspberrypi:~ $ sudo apt-get install python3-edgetpu
Reading package lists... Done
Building dependency tree        
Reading state information... Done
python3-edgetpu is already the newest version (12-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

 

when I try to run one of the examples I get this:

 

pi@raspberrypi:~/Myprojects/Edgetpu/edgetpu-master/examples $ python3 classify_image.py --model models/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --label models/inat_bird_labels.txt --image images/parrot.jpg
Traceback (most recent call last):
  File "classify_image.py", line 17, in <module>
    from edgetpu.classification.engine import ClassificationEngine
ImportError: No module named 'edgetpu.classification'

 

I'm a bit desperate 

 

Regaeds Peter

 

 

 

Gesendet: Samstag, 02. November 2019 um 18:21 Uhr

zubairahmed-ai, 2019-11-04:
@PjotrG 

Did you try my workaround?

PjotrG, 2019-11-04:
@zubairahmed-ai 
Traceback (most recent call last):
  File "/home/pi/Myprojects/Edgetpu/edgetpu-master/examples/classify_capture.py", line 20, in <module>
    from classification.engine import ClassificationEngine
ImportError: No module named classification.engine
yes..like this? failed as well ???????

PjotrG, 2019-11-05:
@zubairahmed-ai 
started everything new with image backup, looks much better but new error showed up, something to do with "loading requirements"???
File "classify_image.py", line 18, in <module>
    from edgetpu.utils import dataset_utils
ImportError: cannot import name 'dataset_utils'

zubairahmed-ai, 2019-11-06:
Yes like that, you have to go in each sub-file that the main script is calling and change it there. Tedious but once done it worked for me.

PjotrG, 2019-11-16:
@zubairahmed-ai @JeanCarm 
Got it all (examples object detection/image classification, teachable machine) working again after installing everything new, looking forward now for learning more and own projects, thank you both for encouragement and help! which projects are you working on? Peter

zubairahmed-ai, 2019-11-16:
@PjotrG I am not working on anything right now, what about you?
I'm looking forward to using it after reading some use cases from [Adrian's upcoming book](https://www.pyimagesearch.com/2019/04/05/table-of-contents-raspberry-pi-for-computer-vision/)

PjotrG, 2019-11-25:
@zubairahmed-ai  I just installed the detect.py (https://github.com/google-coral/examples-camera/tree/master/gstreameron) with minor modifications on my robot arm (OWI 535): the gripper holds the camera and the arm follows a defined object detected (somehow similar to https://github.com/google-coral/project-banana-robo/blob/master/detect_standalone.py) it went "surprinsingly" well.
I now want to extend it to a self driving car model but ran into another installation problem https://github.com/google-coral/examples-camera/issues/19....
Theoretically-wise I got insights from  http://neuralnetworksanddeeplearning.com/index.html ... easy to read but not so easy content. I rebuilt the code on Windows.

jonasl, 2019-11-27:
The python module edgetpu is in the python3-edgetpu package.
sudo apt-get install python3-edgetpu

LifelongLearnerOki, 2020-01-04:
Hi guys! I had the same issue and solved it by Installing the Edge TPU library as follows

```
sudo apt-get update
sudo apt-get install python3-edgetpu
```
Link: https://coral.ai/docs/edgetpu/api-intro

cgillions, 2020-01-06:
I'm able to `import edgetpu` outside of a virtual environment, but inside, I get this error.
I create the virtual environment using `python3 -m venv env` _after_ running `apt-get install python3-edgetpu` but to no avail. Any suggestions?

aeropia, 2020-02-11:
I'm having the same issue in RPi 4B (no virtual env).
Unistall and reinstall does not remove the problem. 
Any solution yet?

jingw222, 2020-02-23:
Hey, 
Installing the Debian package `python3-edgetpu` works fine but only under global system environment. However how can I import and use `edgetpu` in my virtual environment? Is there a pypi release available for the package?

Namburger, 2020-05-21:
Hi guys I'll one up @aeropia. To me this is very standard python issues especially if you have multiple python version install + some odd virtual environment. Check out my answer here https://github.com/robmarkcole/coral-pi-rest-server/issues/56#issuecomment-632125175

closed 2020-05-24 by Namburger