f0cal/google-coral #46: RuntimeError: Error in device opening

habakan, 2019-08-27:
- Doc you were trying to follow: https://coral.withgoogle.com/docs/accelerator/get-started/
- Your Host OS: ubuntu 16.04 tegra (Jetson TX2)
- Your Python3 version: python3.5

# What I Ran

```
$ python3 classify_image.py \
--model ~/Downloads/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
--label ~/Downloads/inat_bird_labels.txt \
--image ~/Downloads/parrot.jpg
```

# What the docs said should happen

You should see results like this:

```
---------------------------
Ara macao (Scarlet Macaw)
Score :  0.761719
```

# What actually happend
```
Traceback (most recent call last):
  File "classify_image.py", line 44, in <module>
    main()
  File "classify_image.py", line 34, in main
    engine = ClassificationEngine(args.model)
  File "/home/nvidia/Downloads/virtualenv/lib/python3.5/site-packages/edgetpu/classification/engine.py", line 47, in __init__
    super().__init__(model_path)
  File "/home/nvidia/Downloads/virtualenv/lib/python3.5/site-packages/edgetpu/basic/basic_engine.py", line 40, in __init__
    self._engine = BasicEnginePythonWrapper.CreateFromFile(model_path)
RuntimeError: Error in device opening (/sys/bus/usb/devices/1-1)!
```

mrusme, 2019-09-05:
@habakan your user is not in the correct set of groups that allow access to the device. Either add your user to the correct group or run as root.

habakan, 2019-09-09:
@mrusme 
Thank you comments.
I ran as root, but the same error happened.

langersteff, 2019-09-10:
Same here unfortunately. Has anyone found a fix yet?

dksim, 2019-09-16:
Try
` sudo usermod -aG plugdev YOURUSERNAME` 
and reboot

habakan, 2019-09-18:
@dksim 
Thank you. 
I tried it but it was same error...
I contacted coral-support team about this problem.
I received a reply "We do not officially support USB Accelerator with Jetson TX2 as of now" .

iatkinson, 2019-10-04:
I'm on the same platform (TX2) and following https://coral.withgoogle.com/docs/edgetpu/api-intro/ (which looks like the same example).  I originally had this same error and after following @dksim am able to run on the USB edgeTPU from Jetson:

$ cd /usr/share/edgetpu/examples/
$ python3 classify_image.py --model models/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --label models/inat_bird_labels.txt --image images/parrot.jpg
---------------------------
Ara macao (Scarlet Macaw)
Score :  0.7265625

habakan, 2019-10-05:
@iatkinson 
Thank you comments!
I confirmed to run by your and @dksim's method!
It seems installation is different from in your provided url.

I installed following script.
```
$ wget https://dl.google.com/coral/edgetpu_api/edgetpu_api_latest.tar.gz -O edgetpu_api.tar.gz --trust-server-names
$ tar xzf edgetpu_api.tar.gz
$ cd edgetpu_api
$ vim install.sh
$ sudo edgetpu_api/install.sh
```

By reinstall using this method, I can run on EdgeTPU Accelarator!

```
echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list

curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -

sudo apt-get update

sudo apt-get install python3-edgetpu
```

Thank you for everyone!

closed 2019-10-05 by habakan