f0cal/google-coral #50: Don't buy a Coral Dev Board - garbage

zorrobyte, 2019-09-27:
Board is being returned.

Installed, installed latest updates. Repeated several times between these issues.

/sys/class/thermal/thermal_zone0/trip_point_4_temp has a temp set of 65000 degrees before the fan turns on. Changes to this file doesn't persist between reboots.

Wifi needs to be manually reconnected every single time.

Compiling C code results in a hard lock after a few seconds of compiling. https://github.com/f0cal/google-coral/issues/49

```
libtool: compile: g++ -DHAVE_CONFIG_H -I. -I./src -I./src -DKJ_HEADER_WARNINGS -DCAPNP_HEADER_WARNINGS -DCAPNP_INCLUDE_DIR="/usr/local/include" -pthread -fPIC -pthread -MT src/capnp/compiler/lexer.lo -MD -MP -MF src/capnp/compiler/.deps/lexer.Tpo -c src/capnp/compiler/lexer.c++ -o src/capnp/compiler/lexer.o >/dev/null 2>&1
[ 726.210412] mmcblk0: error -84 transferring data, sector 2036928, nr 256, cmd response 0x900, card status 0xb00
```

Buyer beware.

zorrobyte, 2019-09-30:
There is a tech assigned to this, so I'll provide an update. I'd like to keep and use the board if possible!
Pardon my frustration, I had high expectations of Google Coral!

Here's a really simple test. Try compiling:
```
curl -O https://capnproto.org/capnproto-c++-0.7.0.tar.gz
tar zxf capnproto-c++-0.7.0.tar.gz
cd capnproto-c++-0.7.0
./configure
make -j6 check
sudo make install
```
https://capnproto.org/install.html

Mine locks up almost immediately!

zorrobyte, 2019-10-01:
- There is only 1GB of RAM. Running out of RAM crashes the board. Allocate a swapfile
- Format and mount an SD card as the onboard 8GB is tiny
- Allocate the swap on the SD so you don't kill the onboard flash
- My wifi issues were due to weak signal. Coral has some weaksauce wifi antennas. Connect via ethernet cable

lleonid, 2019-10-08:
mmcblk0: error -84 transferring data ...
Should be fixed in the upcoming mendel release.

Namburger, 2020-05-24:
@zorrobyte Let me know if anything else I can help with. As Leonid mentioned, most of these issue should be fixed by now, and our apt server is better. As far as OOM causing killed processes, that's quite standard for linux, the solution is usually swap space (pagefile in window). I usually add a 32gb sdcard on my board and use some of the emmc as swap. We are working on new generation with more memories.

closed 2020-05-24 by Namburger