f0cal/google-coral #53: errors broke out during compliation of tflite file for edgetpu

jisooyu, 2019-10-09:
Instructions followed: compilation instructions listed on https://coral.withgoogle.com/docs/edgetpu/retrain-detection/
OS: Host: Mac Mojavee 10.14.8 Local: Ubunbu 18.04 on Virtual Box
Scripts ran : jisooyu@jisooyu-VirtualBox:/usr/bin/edgetpu_compiler_bin$ edgetpu_compiler -s  output_tflite_graph.tflite
Output generated: 
Edge TPU Compiler version 2.0.267685300

Model compiled successfully in 313 ms.

Input model: output_tflite_graph.tflite
Input size: 5.33MiB
Output model: output_tflite_graph_edgetpu.tflite
Output size: 5.75MiB
On-chip memory available for caching model parameters: 7.62MiB
On-chip memory used for caching model parameters: 5.66MiB
Off-chip memory used for streaming uncached model parameters: 0.00B
Number of Edge TPU subgraphs: 1
Total number of operations: 64
Operation log: output_tflite_graph_edgetpu.log

Model successfully compiled but not all operations are supported by the Edge TPU. A percentage of the model will instead run on the CPU, which is slower. If possible, consider updating your model to use only operations supported by the Edge TPU. For details, visit g.co/coral/model-reqs.
Number of operations that will run on Edge TPU: 63
Number of operations that will run on CPU: 1

Operator                       Count      Status

CUSTOM                         1          Operation is working on an unsupported data type
CONCATENATION                  2          Mapped to Edge TPU
CONV_2D                        34         Mapped to Edge TPU
DEPTHWISE_CONV_2D              13         Mapped to Edge TPU
RESHAPE                        13         Mapped to Edge TPU
LOGISTIC                       1          Mapped to Edge TPU
Error opening file for writing: output_tflite_graph_edgetpu.tflite

Anyone please help me to find what is wrong. The error doesn't tell much.

zvit3k, 2019-10-22:
Hi @jisooyu 

As you can see model is correctly compiled. The issue is probably related to the privileges on the command you want to execute.

Try **sudo** edgetpu_compiler output_tflite_graph.tflite

I hope it helps.
Witek

mbcel, 2020-01-08:
Furthermore it seems that as the compilation states that your whole model will not completely run on the edgetpu which probably results in very slow inference time. Dpending on the possition of the unsupported layer.

Ryan-Johnson-1315, 2020-02-10:
I am having the same problem. The model will compile successfully. When I run it on with my TPU it throws these errors:
`Null custom op data
Node number 0 (edgetpu-custom-op) failed to prepare.
`

Any suggestions?

Namburger, 2020-05-24:
@jisooyu You most likely just don't have permission to write in that directory. Moving the model to another directory and compile should fix the problem, or just give yourself permission for the directory with:
```
sudo chown -R $USER:$USER dir_name
```
The problem with compiling with sudo is that it'll most likely write the model as root and then your regular user may not get permission to use the model haha

@Ryan-Johnson-1315 hard to tell without more details but that's most likely because you didn't register the edgetpu ops. Python would looks like this:
```
interpreter = tflite.Interpreter(model_path,
  experimental_delegates=[tflite.load_delegate('libedgetpu.so.1')])
```
C++ would look like this:
```
  tflite::ops::builtin::BuiltinOpResolver resolver;
  resolver.AddCustom(edgetpu::kCustomOp, edgetpu::RegisterCustomOp());
```

closed 2020-05-24 by Namburger