f0cal/google-coral #54: Coral Enviro Sensor Board Stopped working

kflmiami420, 2019-10-10:
Traceback (most recent call last):
  File "/home/pi/enviro_demo-4.py", line 81, in <module>
    main()
  File "/home/pi/enviro_demo-4.py", line 62, in main
    sensors['temperature'] = enviro.temperature * (9.0 / 5.0) + 32
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

I had this line working to convert from C to F but now it says its unsupported

kflmiami420, 2019-10-13:
another error today pi@raspberrypize10:~ $ python3 enviro_demo-4.py
No valid config provided (reading cloud_config.ini).
Cloud IoT is disabled.
Traceback (most recent call last):
  File "enviro_demo-4.py", line 81, in <module>
    main()
  File "enviro_demo-4.py", line 70, in main
    sensors['pressure'] = enviro.pressure * 10.0
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'``

kflmiami420, 2019-10-13:
I comment the * 10 from that line    I dont get a kpa reading , The displays shows nan as a value

Namburger, 2020-05-24:
@kflmiami420 sorry, I can't tell much just from the query, but ususally the nan errors happens when the rpi can't read/write from gpio. We've made lots of changes/bug fixes to the library. Can you check if gpio is turned on with `raspi-config` as well as installing the new library?