f0cal/google-coral #58: Key not present on wishful-yarn -- pushing

rahul4tripathi2, 2019-11-19:
* Doc you were trying to follow: https://coral.withgoogle.com/docs/dev-board/get-started/
* Your host OS: Ubuntu 18.04.3 LTS
* Your Python3 version: python 3.6.8

# What I ran
mdt shell

# What the docs said should happen
it should connect dev board using type-c cable

# What actually happened
rahul@ubuntu:~$ mdt shell
Waiting for a device...
Connecting to wishful-yarn at 192.168.101.2
Key not present on wishful-yarn -- pushing

It looks like you're trying to connect to a device that isn't connected
to your workstation via USB and doesn't have the SSH key this MDT generated.
To connect with `mdt shell` you will need to first connect to your device
ONLY via USB.

Cowardly refusing to attempt to push a key to a public machine.

rahul4tripathi2, 2019-11-20:
Issue Fixed

Steps

1.  Connect your dev board via serial cable and login [Ist terminal].
2.  Once login hit sudo reboot-bootloader to put devboard in fastboot mode.
3. Open 2nd terminal and perform below steps to flash mendel linux

cd $HOME/Downloads

curl -O https://dl.google.com/coral/mendel/enterprise/mendel-enterprise-chef-13.zip

unzip mendel-enterprise-chef-13 \
&& cd mendel-enterprise-chef-13

bash flash.sh

4. Once Flashing the board complete shutdown google coral
sudo shutdown now

5. Connect devboard again with mdt shell and enjoy !!!

rahul@ubuntu:~$ mdt shell
Waiting for a device...
Connecting to wishful-pig at 192.168.100.2
Linux wishful-pig 4.9.51-imx #1 SMP PREEMPT Fri Apr 5 00:17:46 UTC 2019 aarch64

The programs included with the Mendel GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Mendel GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Apr 19 23:02:50 2019 from 192.168.100.68

closed 2019-11-20 by rahul4tripathi2

rahul4tripathi2, 2019-11-27:
correct solution incase you want to stick with new solution
https://stackoverflow.com/questions/58938727/unable-to-connect-google-coral-using-otg-port/59035796#59035796

HughMungis, 2022-11-10:
> 3. Open 2nd terminal and perform below steps to flash mendel linux

I'm using putty and I cannot open a 2nd terminal into the COM port or the alternative COM port that show up in my device manager.

booting into fastboot mode was successful, but i can't get a shell after doing that. Is there some intermediate step i'm missing?