f0cal/google-coral #59: ModuleNotFoundError: No module named 'cv2' in google coral mendel 4.0

rahul4tripathi2, 2019-11-27:
https://coral.withgoogle.com/news/updates-11-2019/

As per above google coral November updates new Mendel 4.0 has the support of OpenCV but when I installed Mendel 4.0 it is unable to import cv2. Please let me know do I've to follow any further steps to install OpenCV in Mendel 4.0

Namburger, 2019-11-27:
@rahul4tripathi2 
```
$ sudo apt-get install python3-opencv
```
should do the trick!

rahul4tripathi2, 2019-11-28:
It is not working!!!

Namburger, 2019-11-28:
@rahul4tripathi2 sorry, are you using python2 or python3?
With python2 it should be
```
$ sudo apt-get install python-opencv
```

rahul4tripathi2, 2019-11-28:
I'm using python3

Namburger, 2019-12-02:
@rahul4tripathi2 after flashing the board, did you connect it to wifi and [update the OS](https://coral.withgoogle.com/docs/dev-board/get-started/#4-update-the-mendel-software)?
```
$ sudo apt-get update
$ sudo apt-get dist-upgrade
```
I just re-flashed my board with mendel day and it had no issue installing it.

rahul4tripathi2, 2020-01-06:
following below link solved my problem
https://stackoverflow.com/questions/59066643/modulenotfounderror-no-module-named-cv2-in-google-coral-mendel-4-0/59077513?noredirect=1#comment104510616_59077513

closed 2020-01-06 by rahul4tripathi2