f0cal/google-coral #60: Parse error in /etc/sudoers near line 28 when doing `sudo apt-get update`

pliablepixels, 2019-12-09:
* Doc you were trying to follow: https://coral.ai/docs/dev-board/get-started/
* Your host OS:  OSX Mojave 10.14.6
* Your Python3 version: 3.6.8 (holy)

# What I ran
```
sudo apt-get update
```

# What the docs said should happen
Should initiate packages updates (step 1 of 2)

# What actually happened
```
mendel@coy-jet:~$ sudo apt-get update
>>> /etc/sudoers: syntax error near line 28 <<<
sudo: parse error in /etc/sudoers near line 28
sudo: no valid sudoers sources found, quitting
sudo: unable to initialize policy plugin
```

pliablepixels, 2019-12-09:
closing - reflashing again worked

closed 2019-12-09 by pliablepixels

gte620v, 2020-12-06:
How did you do the reflashing?  re-entering the bootloader seems to require sudo: https://coral.googlesource.com/mdt/+/refs/heads/master/mdt/shell.py#123

gte620v, 2020-12-06:
Nevermind.  I see you can get back into `fastboot 0` mode by connected to serial before you add USB-C power from the board.  From there, I can enter u-boot and get run `fastboot 0`.