f0cal/google-coral #61: Tutorial: Adding a swap file and moving /home to an SD card

pliablepixels, 2019-12-13:
I thought it might be useful to document the steps to resolve #5 and #7. I'm not an embedded person by any means, so feel free to correct/suggest improvements:

Here is what I did - I added a 128GB micro SD card to the coral board and wanted to do the following:

a) Migrate my existing `/home` to my sd card
b) Set up a 1G swap file, but set a low swapiness value, so its only used when really needed

Caveats:
a) Migrating `/home` and swap to an SD card may reduce the lifespan of the card. Thus it is important to make sure the swap file is only used when really needed. I'm personally not convinced that moving `/home`  to an SD card is bad, but who knows. Don't do it if you don't want to :-)

**Start:**
I assume you have followed the "[Getting Started](https://coral.ai/docs/dev-board/get-started/)"  instructions and are now at a point where you can successfully `mdt shell` into your device and can run the demos the page shows (nice demos).

**Goal:**
You now realize with just 8GB of disk space, you are soon going to run out of space, especially if you want to put in your own python virtual environment with openCV, yolo et al inside it. So you want more disk space. You also realize with just 1GB of memory, you may  run out of memory too for intensive build operations and/or running some models. So just to be sure you want to add a swap, while realizing that swap is not a great idea for an embedded board and it should _only_ be used if you are close to running out of physical memory. You understand that doing all this may reduce your SD card life as well. You say "meh". 

**Steps:**
1. Insert SD card

2. Now get ID of SD card
```
sudo fdisk -l
<various disks, snipped>
Disk /dev/mmcblk1: 116.2 GiB, 124721823744 bytes, 243597312 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
```
So in my case, the SD card is `/dev/mmcblk1`

3. Format disk

```
 sudo mkfs.ext4 /dev/mmcblk1
```

Now wait a while while it formats.

4. Mount newly formatted disk and prepare to migrate existing home

```
# mount it to /mnt 
sudo mount /dev/mmcblk1 /mnt

# copy everything in current /home to /mnt
# command below does not include encrypted directories
sudo rsync -aXS --exclude='/*/.gvfs' /home  /mnt
 
# make sure your copy went will. Diff should not show anything 
# which means everything is same
sudo diff -r /home /mnt/home -x ".gvfs/*"

# make sure new /home/mendel is RW mendel
sudo chown -R mendel:mendel /mnt/home/mendel

```

5. Make the home directory remounting permanent

```
sudo echo " /dev/mmcblk1 /home ext4 defaults 0 1" >> /etc/fstab
```

6. Reboot your machine from an mdt shell and make sure your home directory is all good. Note if you are unable to `mdt shell` you did something wrong. In my case, I forgot to change permissions of the copied directory to `mendel` and also mistakenly moved `/home` to `/home/home` in my SD card with the first rsync command. Worry not, you can always use your serial connector and connect to the coral board - you can login with `mendel/mendel` and rectify things.

7. Assuming now your home directory is migrated and you have validated running the demos etc, lets set up Swap

This sets up a bounded 1G file on your SD card
```
 # setup swap
 sudo fallocate -l 1G /home/swapfile
 sudo chmod 600 /home/swapfile
 sudo mkswap /home/swapfile
 sudo swapon /home/swapfile
```
Make sure its set up correctly:
```
sudo swapon --show
NAME           TYPE  SIZE USED PRIO
/home/swapfile file 1024M   0B   -2
```
Yup.

Now make it permanent
```
sudo echo "/home/swapfile none swap sw 0 0" > /etc/fstab
```

Now make sure you set a [low swappiness](https://linuxhint.com/understanding_vm_swappiness/) to the system doesn't use it aggressively

Edit your ` /etc/sysctl.conf` and add:
```
vm.swappiness=10
```
to the file, if it doesn't already exists. If that value exists, change it to `10`

8. reboot again to make sure all settings are permanent

This is my system after all of that and a fresh reboot

```
l@red-dog:~$ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7294440 1542164   5394784  23% /
devtmpfs          340344       0    340344   0% /dev
tmpfs             504824       0    504824   0% /dev/shm
tmpfs             504824   19388    485436   4% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             504824       0    504824   0% /sys/fs/cgroup
/dev/mmcblk0p1    126931   30370     90008  26% /boot
/dev/mmcblk1   119361328 1115052 112139960   1% /home
tmpfs             504824     428    504396   1% /var/log
tmpfs             100964    4708     96256   5% /run/user/1000

mendel@red-dog:~$ sudo swapon -s
Filename				Type		Size	Used	Priority
/home/swapfile                         	file    	1048572	0	-2

mendel@red-dog:~$ cat /proc/sys/vm/swappiness
10
```

Ecuashungo, 2020-04-23:
I followed your instructions, and somehow messed up something. I wanted to connect through the serial port to rectify things, but I have got the following message:
`Cannot open access to console, the root account is locked.
See sulogin(8) man page for more details.`
Any idea how I can remove the line `/dev/mmcblk1 /home ext4 defaults 0 1` from `/etc/fstab` without serial console?

haithienld, 2020-05-20:
-bash: /etc/fstab: Permission denied
sudo chmod 666 /etc/fstab

Namburger, 2020-05-21:
@pliablepixels Nam from the coral team here, very nice guide, and thanks
Although before making the mkfs.ext4, it's usually standard to create a partition in the sd card first. So something like:
```
sudo fdisk /dev/mmcblk1
 // create new partition
Command (m for help): n
// make this a primary partition
Command (m for help): p
// Write it (by default fdisk will make this a ext4 partition)
Command (m for help): w
```
Then **sudo fdisk -l** should show something like this:
```
....
Disk /dev/mmcblk1: 29.7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x2aaeb742

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk1p1       2048 62333951 62331904 29.7G 83 Linux
```
You'll want to make `/dev/mmcblk1p1` the ext4:
```
sudo mkfs.ext4 /dev/mmcblk1p1
```
And the rest is to follow your tutorial :)

Cheers!

closed 2020-05-24 by Namburger

slbejaranom, 2021-03-01:
I always get to the point where you "permanently mount it" but once i reboot the board, it doesn't mount itself, still don't know why, followed the exact steps verifying not damaging fstab, checked thrice, after reboot the file fstab keeps the modification but the SD doesn't mount itself, am I doing something wrong?

jnnks, 2022-08-07:
> I always get to the point where you "permanently mount it" but once i reboot the board, it doesn't mount itself, still don't know why, followed the exact steps verifying not damaging fstab, checked [...]

If anyone has a similar problem, check your `/etc/fstab` file in a text editor. I already had an entry mounting a partition to `/home` and the one added manually failed because of that.

The partitions are mounted in the order they appear in `/etc/fstab`, so your SD Card should be listed after the original mount:
```bash
/dev/mmcblk0p4 / ext4 noatime,defaults 0 1
/dev/mmcblk0p1 /boot ext2 noatime,defaults 0 2
tmpfs /var/log tmpfs defaults 0 0
# sd card
/dev/mmcblk1p1 /home ext4 defaults 0 1
# default
/dev/mmcblk0p3 /home ext4 noatime,nosuid,nodev,defaults 0 3
```

**Leave the original mount entry in the config file in case something goes wrong.**

brianthelion, 2023-03-29:
@Sruthi-sk Please open a new issue.

nerbivol, 2024-05-22:
> > I always get to the point where you "permanently mount it" but once i reboot the board, it doesn't mount itself, still don't know why, followed the exact steps verifying not damaging fstab, checked thrice, after reboot the file fstab keeps the modification but the SD doesn't mount itself, am I doing something wrong?
> 
> If anyone has a similar problem, check your `/etc/fstab` file in a text editor. I already had an entry mounting a partition to `/home` and the one added manually failed because of that.
> 
> The partitions are mounted in the order they appear in `/etc/fstab`, so your SD Card should be listed after the original mount:
> 
> ```shell
> /dev/mmcblk0p4 / ext4 noatime,defaults 0 1
> /dev/mmcblk0p1 /boot ext2 noatime,defaults 0 2
> tmpfs /var/log tmpfs defaults 0 0
> # sd card
> /dev/mmcblk1p1 /home ext4 defaults 0 1
> # default
> /dev/mmcblk0p3 /home ext4 noatime,nosuid,nodev,defaults 0 3
> ```
> 
> **Leave the original mount entry in the config file in case something goes wrong.**

### Issue Title
Unable to Mount New Partition to /home Directory

### Issue Description
I attempted to add a new partition to the `/home` directory by modifying the `/etc/fstab` file. However, after rebooting the system, the new partition does not seem to be mounted as expected. Below are the steps I followed and the current state of my system.

### Steps to Reproduce
1. Initial state of `/etc/fstab`:
    ```
    PARTUUID=02f36a4f-4562-46b2-bbec-bfc1682f9e92 / ext4 noatime,defaults 0 1
    PARTUUID=b3c678b9-3004-4e30-87dc-ed33fbf4dbfd /home ext4 noatime,nosuid,nodev,defaults,x-systemd.makefs 0 3
    PARTUUID=0f94c91b-1d90-492f-ab86-6952303b214e /boot ext2 noatime,defaults 0 2
    tmpfs /var/log tmpfs defaults 0 0
    ```
2. Output of `df -h` command:
    ```
    Filesystem      Size  Used Avail Use% Mounted on
    /dev/root       5.1G  1.7G  3.1G  36% /
    devtmpfs        967M     0  967M   0% /dev
    tmpfs           999M     0  999M   0% /dev/shm
    tmpfs           999M  8.7M  991M   1% /run
    tmpfs           5.0M  4.0K  5.0M   1% /run/lock
    tmpfs           999M     0  999M   0% /sys/fs/cgroup
    /dev/mmcblk0p2  124M   28M   91M  24% /boot
    /dev/mmcblk0p3  2.0G  1.7G  180M  91% /home
    tmpfs           999M  272K  999M   1% /var/log
    tmpfs           200M  8.3M  192M   5% /run/user/1000
    ```
3. Added the following line to `/etc/fstab`:
    ```
    /dev/mmcblk2p1 /home ext4 defaults 0 1
    ```
4. New state of `/etc/fstab` after modification:
    ```
    PARTUUID=02f36a4f-4562-46b2-bbec-bfc1682f9e92 / ext4 noatime,defaults 0 1
    PARTUUID=b3c678b9-3004-4e30-87dc-ed33fbf4dbfd /home ext4 noatime,nosuid,nodev,defaults,x-systemd.makefs 0 3
    PARTUUID=0f94c91b-1d90-492f-ab86-6952303b214e /boot ext2 noatime,defaults 0 2
    tmpfs /var/log tmpfs defaults 0 0
    /dev/mmcblk2p1 /home ext4 defaults 0 1
    ```
5. Rebooted the system.

### Expected Behavior
The new partition `/dev/mmcblk2p1` should be mounted to the `/home` directory.

### Actual Behavior
The `/home` directory remains on the original partition:
```
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       5.1G  1.7G  3.1G  36% /
devtmpfs        967M     0  967M   0% /dev
tmpfs           999M     0  999M   0% /dev/shm
tmpfs           999M  8.7M  991M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           999M     0  999M   0% /sys/fs/cgroup
/dev/mmcblk0p2  124M   28M   91M  24% /boot
/dev/mmcblk0p3  2.0G  1.7G  180M  91% /home
tmpfs           999M  272K  999M   1% /var/log
tmpfs           200M  8.3M  192M   5% /run/user/1000
```

### Possible Suggested Fix
Modification for `/etc/fstab`:
```
    PARTUUID=02f36a4f-4562-46b2-bbec-bfc1682f9e92 / ext4 noatime,defaults 0 1
    /dev/mmcblk2p1 /home ext4 defaults 0 1
    PARTUUID=b3c678b9-3004-4e30-87dc-ed33fbf4dbfd /home ext4 noatime,nosuid,nodev,defaults,x-systemd.makefs 0 3
    PARTUUID=0f94c91b-1d90-492f-ab86-6952303b214e /boot ext2 noatime,defaults 0 2
    tmpfs /var/log tmpfs defaults 0 0
```

Is it supposed to work?