f0cal/google-coral #63: Google Coral Dev Board cannot connect to MDT  shell

vkolli3, 2019-12-23:
I followed getting-started guide of the Coral Dev Board  provided by Google: Get started with the Dev Board then I reached the title "Connect to the board's shell via MDT" but I received the following error message:

~$ mdt shell
Connecting to neat-llama at 192.168.101.2
Key not present on 192.168.101.2 -- pushing

It looks like you're trying to connect to a device that isn't connected
to your workstation via USB and doesn't have the SSH key this MDT generated.
To connect with `mdt shell` you will need to first connect to your device
ONLY via USB.

Cowardly refusing to attempt to push a key to a public machine.

I connected the Devboard using USB-C data port labeled "OTG" on the Dev Board. I am using Ubuntu 18.04.3 LTS OS on my host computer.Also, the device is successfully detected with mdt devices command,

vkolli3, 2019-12-23:
I resolved it by using the solution proposed by rahul4tripathi2 at
https://github.com/f0cal/google-coral/issues/58.

Thank You Rahul

closed 2019-12-23 by vkolli3

haithienld, 2021-03-02:
You need to check your board. If your board is connecting to the Internet throught wifi. And you want to use mdt shell, you should to disable the connection. Then, connect your board to host through OTG then use mdt shell.