f0cal/google-coral #64: how to set auto-start a application？I can not find rc.local in /etc .

dinghuawen, 2020-01-02:
* Doc you were trying to follow: <please insert>
* Your host OS: <please insert>
* Your Python3 version: <please insert>

# What I ran

# What the docs said should happen

# What actually happened

aaroninaustin, 2020-01-22:
So you'll want to use `systemd` - I hadn't done it in a while myself and followed this guide:

https://www.digitalocean.com/community/tutorials/how-to-use-systemctl-to-manage-systemd-services-and-units

My only tips to add would be that actually create the services file you'll edit a new one and get a blank file to drop the service config in...

`sudo systemctl edit yourapp.service`

From that point you can start with something like this to get a service initialized. After which you'll
enable it for startup `sudo systemctl enable yourapp.service`

```
[Unit]
Description=Your App
After=network.target
StartLimitIntervalSec=0

[Service]
Type=simple
Restart=always
RestartSec=1
User=mendel
ExecStart=/usr/bin/bash /home/mendel/script.sh

[Install]
WantedBy=multi-user.target
```

Namburger, 2020-05-24:
Honorable mention: https://github.com/google-coral/edgetpu/issues/120#issuecomment-630814040

closed 2020-05-24 by Namburger