f0cal/google-coral #65: How to quantify a trained model without its original database?

coollofty, 2020-01-15:
I have a trained .tflite model file, I want to run this model on a coral usb , so I need to quantify it as an Int8 type but the original file is gone. How do I accomplish this? Thanks!

Namburger, 2020-05-21:
@coollofty could you clarify what you mean by the original file is gone?

coollofty, 2020-05-22:
> could you clarify what you mean by the original file is gone

original file is the trainned .ptb file, quantification .ptb to .tflite int8 model. If I can't find the .ptb file, can I still implement this quantification process

Namburger, 2020-05-22:
@coollofty Ahh, I see now. 
Hummn, I don't know if this is possible, unfortunately. The tflite converter were meant for graph -> tflite only, I've attempted tflite -> tflite. Maybe the tensorflow team can give you a better answer for this question though.

closed 2020-05-24 by Namburger