f0cal/google-coral #66: Julia language with Coral board

RBAM, 2020-01-15:
Hi

I am doing a research where I have to do a benchmark between a PC with GPU vs the Coral board’s TPU. And I have to use Julia.

I need some guidance. For example, I have found TensorFlow.jl (https://github.com/malmaud/TensorFlow.jl). But  do not know how it will help me as the Coral board uses TensorFlow Lite. And I am just starting with it.

Any help on how to get started will be appreciated.

RBAM, 2020-01-16:
[image]

My plan is to use TensorFlow.jl to create the models. Then I will convert them into TensorFlow Lite.

I should work as in this video: https://www.youtube.com/watch?v=n2MwJ1guGVQ  (minute 4 with 13 seconds); the presenter says that it is possible to work with people who use Python or Julia.

Namburger, 2020-05-21:
Hi there, regarding installing `julia` on the dev board:
```
wget https://julialang-s3.julialang.org/bin/linux/aarch64/1.4/julia-1.4.1-linux-aarch64.tar.gz
tar xvf julia-1.4.1-linux-aarch64.tar.gz
./julia-1.4.1/bin/julia
```

Although training and creating models on the dev board is probably not a great idea. Model pipelining is usually done in a robust host machine. Model inference is what's the dev board was designed for.

closed 2020-05-24 by Namburger