f0cal/google-coral #67: Track a specific person/object

GOBish, 2020-01-19:
Hi!
I am using coral in a robot application and was hoping someone could point me in the right direction.
I have a tensor flow model working well, performance is great.
I am working on a person following project, where I would like to designate one person to follow. This can be done manually (don’t need facial recognition or anything like that). Then I would like to have the robot “lock on” to that person so that it can follow him or her.
Any recommendations for a good way to do this?
Thanks!

brianthelion, 2020-01-19:
@GOBish Can you clarify which view of the person the robot will follow? If the tracking algorithm is "designated" to a frontal view of the human and meant to track that same view, then any classic tracking algorithm will do a reasonable job assuming minimal occlusion. If, however, you want to designate to the frontal view and then track the target from behind (as would be the case with a following robot) then you'll need something more sophisticated. Please advise.

GOBish, 2020-01-19:
Thank for the reply @brianthelion ! Would be tracking the rear view of the person.

brianthelion, 2020-02-09:
@GOBish If you can designate the rear view of the person, and the scene is relatively uncluttered -- ie, no crowds of people, no major occlusions -- then OpenCV tracking will do fine to start.

GOBish, 2020-02-09:
What tracker are you using?

Namburger, 2020-05-21:
@GOBish Hi there, you can try working form this object tracking project form the automl team it may give you a better idea:
https://github.com/google/automl-video-ondevice#for-coral-device

GOBish, 2020-05-21:
Ok - I am training a tflite model now and converting it for use on the EdgeTPU. I will see how the automl tracking works.