f0cal/google-coral #68: Can't install scipy module on Coral Dev Board

yieniggu, 2020-01-31:
I'm trying to install scipy module with:

`pip install scipy `

and this is the output I get:
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /home/mendel/.local/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-hyfofxfv/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- wheel setuptools 'Cython>=0.29.13' 'numpy==1.13.3; python_version=='"'"'3.5'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.13.3; python_version=='"'"'3.6'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.17.3; python_version>='"'"'3.8'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.5'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.6'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.7'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.17.3; python_version>='"'"'3.8'"'"' and platform_system=='"'"'AIX'"'"'' 'pybind11>=2.4.0'
       cwd: None
  Complete output (1059 lines):
  Ignoring numpy: markers 'python_version == "3.5" and platform_system != "AIX"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.6" and platform_system != "AIX"' don't match your environment
  Ignoring numpy: markers 'python_version >= "3.8" and platform_system != "AIX"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.5" and platform_system == "AIX"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.6" and platform_system == "AIX"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.7" and platform_system == "AIX"' don't match your environment
  Ignoring numpy: markers 'python_version >= "3.8" and platform_system == "AIX"' don't match your environment
  Collecting wheel
    Using cached wheel-0.34.2-py2.py3-none-any.whl (26 kB)
  Collecting setuptools
    Using cached setuptools-45.1.0-py3-none-any.whl (583 kB)
  Collecting Cython>=0.29.13
    Using cached Cython-0.29.14.tar.gz (2.1 MB)
  Collecting numpy==1.14.5
    Using cached numpy-1.14.5.zip (4.9 MB)
  Collecting pybind11>=2.4.0
    Using cached pybind11-2.4.3-py2.py3-none-any.whl (150 kB)
  Building wheels for collected packages: Cython, numpy
    Building wheel for Cython (setup.py): started
    Building wheel for Cython (setup.py): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-t3imqegd/Cython/setup.py'"'"'; __file__='"'"'/tmp/pip-install-t3imqegd/Cython/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-zedm4q3n
         cwd: /tmp/pip-install-t3imqegd/Cython/
    Complete output (333 lines):
    Unable to find pgen, not compiling formal grammar.
    running bdist_wheel
    running build
    running build_py
    creating build
    creating build/lib.linux-aarch64-3.7
    copying cython.py -> build/lib.linux-aarch64-3.7
    creating build/lib.linux-aarch64-3.7/Cython
    copying Cython/CodeWriter.py -> build/lib.linux-aarch64-3.7/Cython
    copying Cython/StringIOTree.py -> build/lib.linux-aarch64-3.7/Cython
    copying Cython/Coverage.py -> build/lib.linux-aarch64-3.7/Cython
    copying Cython/Shadow.py -> build/lib.linux-aarch64-3.7/Cython
    copying Cython/Utils.py -> build/lib.linux-aarch64-3.7/Cython
    copying Cython/__init__.py -> build/lib.linux-aarch64-3.7/Cython
    copying Cython/TestUtils.py -> build/lib.linux-aarch64-3.7/Cython
    copying Cython/Debugging.py -> build/lib.linux-aarch64-3.7/Cython
    creating build/lib.linux-aarch64-3.7/Cython/Build
    copying Cython/Build/IpythonMagic.py -> build/lib.linux-aarch64-3.7/Cython/Build
    copying Cython/Build/Cythonize.py -> build/lib.linux-aarch64-3.7/Cython/Build
    copying Cython/Build/Distutils.py -> build/lib.linux-aarch64-3.7/Cython/Build
    copying Cython/Build/Inline.py -> build/lib.linux-aarch64-3.7/Cython/Build
    copying Cython/Build/Dependencies.py -> build/lib.linux-aarch64-3.7/Cython/Build
    copying Cython/Build/__init__.py -> build/lib.linux-aarch64-3.7/Cython/Build
    copying Cython/Build/BuildExecutable.py -> build/lib.linux-aarch64-3.7/Cython/Build
    creating build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/Builtin.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/DebugFlags.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/Naming.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/Errors.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/Lexicon.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/Annotate.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/FlowControl.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/StringEncoding.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/Visitor.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/CmdLine.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/MemoryView.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
    copying Cython/Compiler/TreePath.py -> build/lib.linux-aarch64-3.7/Cython/Compiler
[... 977 lines cut ...]
      copying Cython/Utility/CythonFunction.c -> build/lib.linux-aarch64-3.7/Cython/Utility
      copying Cython/Utility/Coroutine.c -> build/lib.linux-aarch64-3.7/Cython/Utility
      copying Cython/Utility/Overflow.c -> build/lib.linux-aarch64-3.7/Cython/Utility
      copying Cython/Utility/arrayarray.h -> build/lib.linux-aarch64-3.7/Cython/Utility
      copying Cython/Utility/CppSupport.cpp -> build/lib.linux-aarch64-3.7/Cython/Utility
      running build_ext
      building 'Cython.Plex.Scanners' extension
      creating build/temp.linux-aarch64-3.7
      creating build/temp.linux-aarch64-3.7/tmp
      creating build/temp.linux-aarch64-3.7/tmp/pip-install-t3imqegd
      creating build/temp.linux-aarch64-3.7/tmp/pip-install-t3imqegd/Cython
      creating build/temp.linux-aarch64-3.7/tmp/pip-install-t3imqegd/Cython/Cython
      creating build/temp.linux-aarch64-3.7/tmp/pip-install-t3imqegd/Cython/Cython/Plex
      aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.7m -c /tmp/pip-install-t3imqegd/Cython/Cython/Plex/Scanners.c -o build/temp.linux-aarch64-3.7/tmp/pip-install-t3imqegd/Cython/Cython/Plex/Scanners.o
      /tmp/pip-install-t3imqegd/Cython/Cython/Plex/Scanners.c:19:10: fatal error: Python.h: No such file or directory
       #include "Python.h"
                ^~~~~~~~~~
      compilation terminated.
      error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
      ----------------------------------------
  ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-t3imqegd/Cython/setup.py'"'"'; __file__='"'"'/tmp/pip-install-t3imqegd/Cython/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-and88dwi/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-hyfofxfv/overlay --compile --install-headers /tmp/pip-build-env-hyfofxfv/overlay/include/python3.7m/Cython Check the logs for full command output.
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /home/mendel/.local/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-hyfofxfv/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- wheel setuptools 'Cython>=0.29.13' 'numpy==1.13.3; python_version=='"'"'3.5'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.13.3; python_version=='"'"'3.6'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"' and platform_system!='"'"'AIX [... cut ...] =1.17.3; python_version>='"'"'3.8'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.5'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.6'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.7'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.17.3; python_version>='"'"'3.8'"'"' and platform_system=='"'"'AIX'"'"'' 'pybind11>=2.4.0' Check the logs for full command output.

Any help would be apreciated, Regards.

brianthelion, 2020-01-31:
`sudo apt-get install python-dev` as indicated by the error message. If you want software-accelerated numerical operations, it's telling you that there are some other things for you to install as well.

Learning to read the Python installer output and figure out what system packages are missing is a bit of an art, but please try before posting.

yieniggu, 2020-01-31:
I tried that and didn't work, looks like you're a pretty smart artist.

`sudo apt-get install python-numpy` did the work :)

closed 2020-01-31 by brianthelion