f0cal/google-coral #69: Deeplabv3 cityscape conversion

Valdiolus, 2020-02-04:
Hi! 
I am looking for deeplabv3 model with cityscape pretrained, optimized for edgetpu (edgetpu.tflite) with max fps rate. 
Trying to convert from "mobilenetv3_small_cityscapes_trainfine" .pb file by myself, but no luck - dimensions of the input of the model is surprisingly 1029*2050, and issues like this, so it doesn't work:
Number of operations that will run on Edge TPU: 2
Number of operations that will run on CPU: 124

I have tried to convert ->.tflite -> edgetpu.tflite with "mobilenetv2_dm05_coco_voc_trainaug_8bit" from here [https://github.com/tensorflow/models/blob/master/research/deeplab/g3doc/quantize.md](url) and it converts great! 

Can you help me with how to get deeplabv3 cityscape model 8bit?

Thank you!

Namburger, 2020-05-21:
What command did you use to converts the deeplabv3 model?

Tetsujinfr, 2020-10-14:
Hi @Valdiolus 
Did you manage to convert deeplabv3 for cityscape? If how did you fix your issue above?
Thanks

Tetsujinfr, 2020-10-14:
Thanks for your note and managing my expectations as well.  If you find the code that that would be great.
When you say the results were very poor you mean the quality of the segmentation, the inference performance, both?

I find the quality of 513x513 segmantation of deeplabv3 on pascal voc 2012 not too bad given the small size of the model (although admittingly it is not great neither), and it runs at approx 7fps on my hardware setup which is a bit slow but manageable. Hence why I had hopes for Cityscape...

Valdiolus, 2020-10-15:
The accuracy of the segmentation was bad, needed more deeply retrain the nn. Speed was quite good, but I don't remember exact fps rate.

Tetsujinfr, 2020-10-15:
Have you tried other sementic segmentation models for outdoor/road/car classes on the edgetpu? 
E.g. I found [this](https://tfhub.dev/sayakpaul/lite-model/deeplabv3-mobilenetv2-ade20k/1/default/2) tflite small model trained on ADE20K which I thought would be a good candidate for the edgetpu, but it seems it is using float32 weights and I can not convert it to run on the coral accelerator properly. (also I am not sure of its accuracy, when I played with the collab notebook demo, the accuracy after inference was not very good, and that was not even with 8bits quantization)

Namburger, 2020-10-15:
Hi forks - 
Sorry, just wanted to give you guys an update here, as @Valdiolus mentioned, the tflite segmentation model trained with cityscape has terrible accuracy in our test rack. This is very surprising to us and we found this out last month. Currently we are sending this to the model team to take a look.

Tetsujinfr, 2020-10-15:
Ok thanks for the update. Looking fwd for a fix if possible on this.
About the conversion process, is there a documentation or tutorial on how to prep a .pb model file to tflite then edgetpu properly?
I came across various materials on the web (most from google/TFlow teams), but I keep getting errors when trying to convert models, even when those are advertized as "edgetpu" specific like the ADE20K above. Not sure if I am doing something stupid but can not figure out what to do.
Can share my conversion code if that helps.

Tetsujinfr, 2020-10-27:
hi @Namburger 
do you have any update regarding the segmentation model accuracy review by your engineering team? do you have a rough timeline on when this might be re mediated?
thanks

Tetsujinfr, 2020-11-09:
hi, any update on the segmentation model accuracy fix on edgetpu? thanks

Tetsujinfr, 2020-11-30:
Hi
Is there any update on the segmentation model accuracy on coral?
Thanks

Tetsujinfr, 2020-12-13:
hi, @Namburger , do you have any update to share on this model accuracy improvement for the edge tpu? thank you