f0cal/google-coral #70: mdt devices doesn't return anything on MacOS X Catalina

iitgrad, 2020-02-04:
* Doc you were trying to follow: Basic setup instructions from google.
* Your host OS: Mac OS X Catalina
* Your Python3 version: 3.73

# 
mdt devices

# What the docs said should happen
a device should be listed

# What actually happened
no devices found

Prem95, 2020-02-12:
There are some things you could try to solve this:

1) run `kextstat`
You must be able to see this below. If not, enable it [Here]. (https://www.silabs.com/community/interface/forum.topic.html/unable_to_installvc-MWVe)

` 216    0 0xffffff7f843db000 0x9000     0x9000     com.silabs.driver.CP210xVCPDriver (5.2.4) 79BD054F-7A71-34DE-9763-59D44DAC0615 <86 27 5 3>`

2) Have you tried running `screen /dev/ttyUSB`? What does it output? If assuming the driver is loaded from (1)

3) Are you able to login to the coral using `fastboot`? If yes, connect to a network. It has to be the same network your host machine is connected to. Once connected, just to check, run `nmtui`
if you see the same network, on your host machine (MAC), type `mdt devices`

Keep trying. Thank you.

fjp, 2020-02-17:
I have the same problem. I found [this blog post](http://blog.toelt.ai/coral/2019/12/18/coral_setup_on_catalina.html) which says:

> Problem is that on Catalina connecting the USB-C cable does not do anything. Not on my laptop and according to several posts I found also for many other people. If you have a Linux machine is no problem, not on Catalina. So you cannot activate ssh as the instructions explain. This may change in the future so just in case, as I am writing, I have tested this with Catalina 10.15.1.

The proposed solution in the blog post is to allowing ssh with only password (which is not safe). 

So far I was able to connect to the host machine (macOS with Catalina) and the Coral dev board to the same wifi network (For this I logged into the serial console via screen and used nmtui to select the wifi network). 

After this I ran mdt devices which listed the dev board. However, this led to the following problem:

```
mdt pushkey mdt.key 
Waiting for a device...
Connecting to jumping-shrimp at 192.168.178.51
Pushing mdt.key
Key not present on jumping-shrimp -- pushing

Couldn't connect to keymaster on jumping-shrimp: [Errno 61] Connection refused.

Did you previously connect from a different machine? If so,
mdt-keymaster will not be running as it only accepts a single key.

You will need to either:
   1) Remove the key from /home/mendel/.ssh/authorized_keys on the
      device via the serial console

- or -

   2) Copy the mdt private key from your home directory on this host
      in ~/.config/mdt/keys/mdt.key to the first machine and use
      'mdt pushkey mdt.key' to add that key to the device's
      authorized_keys file.

Failed to push via keymaster -- will attempt password login as a fallback.
Can't login using default credentials: Bad authentication type; allowed types: ['publickey']
```

Note that I also tried [this patch](https://stackoverflow.com/questions/58938727/unable-to-connect-google-coral-using-otg-port/59035796#59035796):

> 1 Open the file in your favourite text editor
> ```
> vim $HOME/.local/lib/python3.6/site-packages/mdt/sshclient.py +86
> ```
> 2. Replace line 86, which is
> ```
>     if not self.address.startswith('192.168.100'):
>         raise NonLocalDeviceError()
> ```
> by the following code
> ```
>     if not self.address.startswith('192.168.10'):
>         raise NonLocalDeviceError()
> ```

Instead of 192.168.10 I used 192.168.1 because of my DHCP IP Adresses 192.168.178.xx

Namburger, 2020-05-21:
@fjp 

@fjp 
The problem with Catalina is very odd, and I believe it's due to the OS it self not having support for libusb, I don't have a mac to test my self, but here's a safer solution or standard solution is to create a id_rsa key on your mac with:
```
$ ssh-keygen
```
This creates a `~/.ssh/id_rsa.pub` file on your MAC. You can copy that key over to the board and put the content in `~/.ssh/authorized_keys` (you can put multiple keys in this file). 
After that you can go ahead and disable Password authentication but modifying `/etc/ssh/sshd_config` and turn `PasswordAuthentication` from no to yes and restart ssh service on the dev board:
```
$ sudo systemctl restart ssh
```
After that ssh-ing into the dev board will use key authentication instead of password :)

valiantljk, 2020-06-08:
I tested on Catalina 10.15.5 as of today, seeing the same issue. Considering switch to Linux.

DataScience-RajKapoor, 2020-06-11:
@valiantljk 

I was seeing the same issue of missing devices- however, after different tangential experiments,  "mdt devices" started showing up the dev board on macOS Catalina 10.15.5 

The directions that I followed were the ones mentioned in this issue log (

1) At the stage where you connect to the device via mdt , reconnect the serial port
2) log in into dev board via mendel/mendel username and password ( after screen 

After this step , I did couple of different actions - some permutation worked in my favor 

3 a) As shared by @Namburger , I copied over the key and created a file in dev board (you   
       would have logged in coral dev board in step 2)
 
  b) I made the change to /etc/ssh/sshd_config as per his post 
       However this step gets overwritten at later stage in Step 5

  c) I then restarted the `sudo  systemctl restart ssh`

4) I also had my data cable connected to coral and computer 

5) I then configured wireless connection on the dev ( while logged into the dev board as step 2)  
     following the instructions on the official coral (https://coral.ai/docs/dev-board/get-started/#3-connect-to-the-internet)

6) Update the distro as per the official instructions 

7) You will be prompted to either accept the new `sushi_config` file that was changed in 3 b)
     I did accept the new file hoping that some configurations may fix things.

8) After all the updates are finished , on the other terminal, type in `mdt devices` and voila - it showed up the coral dev board 

Hope it helps

Nanotorre, 2020-06-18:
Same problem here with mac os Catalina 10.15.5

raghavendravp, 2020-07-31:
> @iitgrad @fjp
> The problem with Catalina is very odd, and I believe it's due to the OS it self not having support for libusb, I don't have a mac to test my self, but here's a safer solution or standard solution is to create a id_rsa key on your mac with:
> 
> ```
> $ ssh-keygen
> ```
> 
> This creates a `~/.ssh/id_rsa.pub` file on your MAC. You can copy that key over to the board and put the content in `~/.ssh/authorized_keys` (you can put multiple keys in this file).
> After that you can go ahead and disable Password authentication by modifying `/etc/ssh/sshd_config` and turn `PasswordAuthentication` from no to yes and restart ssh service on the dev board:
> 
> ```
> $ sudo systemctl restart ssh
> ```
> 
> After that ssh-ing into the dev board will use key authentication instead of password :)

Having similar troubles. I wonder if you could help me more. I am using Catalina OS as well.
I see the device after running "mdt devices" but I am seeing the same errors. I followed all directions until the last step that you indicated, and when tried to login back using "mdt shell", after the change on the dev board's sshd_config I see a new set of errors. Could you please help provide a resolution?

Waiting for a device...
Connecting to bored-panda at 192.168.xx.xxx
Key not present on bored-panda -- pushing

Couldn't connect to keymaster on bored-panda: [Errno 61] Connection refused.

Did you previously connect from a different machine? If so,
mdt-keymaster will not be running as it only accepts a single key.

You will need to either:
   1) Remove the key from /home/mendel/.ssh/authorized_keys on the
      device via the serial console

- or -

   2) Copy the mdt private key from your home directory on this host
      in ~/.config/mdt/keys/mdt.key to the first machine and use
      'mdt pushkey mdt.key' to add that key to the device's
      authorized_keys file.

Failed to push via keymaster -- will attempt password login as a fallback.
Traceback (most recent call last):
  File "/Users/xxxxxx/Library/Python/3.8/lib/python/site-packages/mdt/sshclient.py", line 92, in _pushKeyViaKeymaster
    connection.request('PUT', '/', key_line + '\r\n')
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/http/client.py", line 1255, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/http/client.py", line 1301, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/http/client.py", line 1250, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/http/client.py", line 1010, in _send_output
    self.send(msg)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/http/client.py", line 950, in send
    self.connect()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/http/client.py", line 921, in connect
    self.sock = self._create_connection(
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 61] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/xxxxxx/Library/Python/3.8/lib/python/site-packages/mdt/sshclient.py", line 138, in pushKey
    self._pushKeyViaKeymaster()
  File "/Users/xxxxxx/Library/Python/3.8/lib/python/site-packages/mdt/sshclient.py", line 109, in _pushKeyViaKeymaster
    raise KeyPushError(e)
mdt.sshclient.KeyPushError: [Errno 61] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./mdt", line 8, in <module>
    sys.exit(main())
  File "/Users/xxxxxx/Library/Python/3.8/lib/python/site-packages/mdt/main.py", line 162, in main
    exit(command.run(sys.argv[1:]))
  File "/Users/xxxxxx/Library/Python/3.8/lib/python/site-packages/mdt/command.py", line 76, in run
    return self.runWithClient(client, args)
  File "/Users/xxxxxx/Library/Python/3.8/lib/python/site-packages/mdt/shell.py", line 74, in runWithClient
    channel = client.openShell()
  File "/Users/xxxxxx/Library/Python/3.8/lib/python/site-packages/mdt/sshclient.py", line 187, in openShell
    self.pushKey()
  File "/Users/xxxxxx/Library/Python/3.8/lib/python/site-packages/mdt/sshclient.py", line 141, in pushKey
    self._pushKeyViaDefaultLogin()
  File "/Users/xxxxxx/Library/Python/3.8/lib/python/site-packages/mdt/sshclient.py", line 129, in _pushKeyViaDefaultLogin
    key_line = self._generateAuthorizedKeysLine()
AttributeError: 'SshClient' object has no attribute '_generateAuthorizedKeysLine'

Namburger, 2020-07-31:
@raghavendravp After following the steps here https://github.com/f0cal/google-coral/issues/70#issuecomment-632102704
you'll need to use ssh instead of mdt:
```
ssh mendel@192.168.100.2
```
you can use that address if the board is connected to your host via usb-c, or if your board is on the same wifi, you can use the wlan0 address :)

raghavendravp, 2020-08-01:
> @raghavendravp After following the steps here [#70 (comment)](https://github.com/f0cal/google-coral/issues/70#issuecomment-632102704)
> you'll need to use ssh instead of mdt:
> 
> ```
> ssh mendel@192.168.100.2
> ```
> 
> you can use that address if the board is connected to your host via usb-c, or if your board is on the same wifi, you can use the wlan0 address :)

Of course! Thank you.. I had an IP address already as I was able to get the device connected to Wlan0. I will just not be able to use mdt, which is fine, I can rely on SSH to perform all my operations.