f0cal/google-coral #72: Error while flashing leading to unresponsive board

Ailsorxd, 2020-02-18:
I was flashing the dev board according to the instructions on the coral website when it got stuck for a while with the terminal saying waiting for device to reboot. After leaving it for 10 minutes with no movement I disconnected it to attempt to start installation again but now when I try to install it, it gets stuck at the screen step.. it shows a blank screen with no U-Boot prompt at all.

When I try the tutorial for reflashing the dev board I get stuck at the mdt devices command which doesn't show any device. What can I do to bring it back to factory defaults or get it to run again?

zerocool95, 2020-02-26:
I had the same problem on mac os. It worked after a system restart. Not sure what the root cause was.

Ailsorxd, 2020-02-26:
I tried everything to get the U-Boot prompt to show up on screen or to make the board appear on mdt.. nothing worked.
Finally what worked was following the tutorial section which showed a way to boot it using the mirco SD card.

closed 2020-02-26 by Ailsorxd