f0cal/google-coral #74: mdt shell failed

jiapei100, 2020-04-07:
* Doc you were trying to follow: [https://coral.ai/docs/dev-board/mdt/](https://coral.ai/docs/dev-board/mdt/)
* Your host OS: Ubuntu 18.04.4
* Your Python3 version: 3.6.9

# What I ran

```console
➜  mendel-enterprise-day-13 mdt devices
mocha-shrimp		(192.168.100.2)
➜  mendel-enterprise-day-13 ping 192.168.100.2
PING 192.168.100.2 (192.168.100.2) 56(84) bytes of data.
64 bytes from 192.168.100.2: icmp_seq=1 ttl=64 time=0.985 ms
64 bytes from 192.168.100.2: icmp_seq=2 ttl=64 time=0.847 ms
64 bytes from 192.168.100.2: icmp_seq=3 ttl=64 time=4.60 ms
^C
--- 192.168.100.2 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2002ms
rtt min/avg/max/mdev = 0.847/2.146/4.607/1.741 ms
➜  mendel-enterprise-day-13 mdt shell
Waiting for a device...
Connecting to mocha-shrimp at 192.168.100.2
Socket exception: Connection reset by peer (104)
Couldn't establish ssh connection to device: socket error: [Errno 104] Connection reset by peer
```

Since I've already manually setup a STATIC IP from within mendel OS via **screen** as **192.168.1.97**, I also tried to load **mdt shell** using its IP address.

```console
➜  mendel-enterprise-day-13 mdt shell 192.168.1.97
Connecting to 192.168.1.97 at 192.168.1.97
Traceback (most recent call last):
  File "~/.local/bin/mdt", line 8, in <module>
    sys.exit(main())
  File "~/.local/lib/python3.6/site-packages/mdt/main.py", line 162, in main
    exit(command.run(sys.argv[1:]))
  File "~/.local/lib/python3.6/site-packages/mdt/command.py", line 76, in run
    return self.runWithClient(client, args)
  File "~/.local/lib/python3.6/site-packages/mdt/shell.py", line 74, in runWithClient
    channel = client.openShell()
  File "~/.local/lib/python3.6/site-packages/mdt/sshclient.py", line 185, in openShell
    if self.shouldPushKey():
  File "~/.local/lib/python3.6/site-packages/mdt/sshclient.py", line 74, in shouldPushKey
    compress=True)
  File "~/.local/lib/python3.6/site-packages/paramiko/client.py", line 406, in connect
    t.start_client(timeout=timeout)
  File "~/.local/lib/python3.6/site-packages/paramiko/transport.py", line 660, in start_client
    raise e
  File "~/.local/lib/python3.6/site-packages/paramiko/transport.py", line 2055, in run
    ptype, m = self.packetizer.read_message()
  File "~/.local/lib/python3.6/site-packages/paramiko/packet.py", line 459, in read_message
    header = self.read_all(self.__block_size_in, check_rekey=True)
  File "~/.local/lib/python3.6/site-packages/paramiko/packet.py", line 303, in read_all
    raise EOFError()
EOFError
```

There are several potential bugs:

1. [https://stackoverflow.com/questions/58938727/unable-to-connect-google-coral-using-otg-port](https://stackoverflow.com/questions/58938727/unable-to-connect-google-coral-using-otg-port)

2. [https://github.com/paramiko/paramiko/issues/1181](https://github.com/paramiko/paramiko/issues/1181)

Can anybody help please?

Cheers
Pei

Namburger, 2020-05-21:
@jiapei100
Can you try `stracing` the process to see what's causing the segfault.

`mdt` is just a user friendly wrapper around ssh so it may be a better solution to just shh into the board.

Charlygnusa, 2020-08-27:
Hello, I have the same problem, is there a solution?

I just followed the documentation for the steps "Execute the flash script", then tried to connect to the board via MDT and SSH, but it didn't work, the same error appeared as above ..

Please help, thanks in advance

Namburger, 2020-08-27:
@Charlygnusa what OS are you using and how have you tried to ssh?

Charlygnusa, 2020-08-27:
@Namburger thank you for the reply

My OS is Ubuntu 18.04, and my python3 is version 3.6.9, I tried ssh like this `ssh vexing-goose@192.168.100.2` but can't, the error is displayed `Connection reset by 192.168.100.2 port 22`

Is it not possible to SSH because I have never connected via MDT?
because in the documentation there is also `You must connect to the board via MDT over USB at least once before you can SSH via any other method.`, but I tried to the connect board via MDT, still it couldn't

Namburger, 2020-08-27:
@Charlygnusa no, you should be able to ssh, although that's not the correct usage + mendel won't allow password authentication by default. Here are the steps:
1) screen into the board's serial console: `screen /dev/ttyUSB0 115200` (make sure serial cable are connected, user/password are both "mendel")
2) open the file `/etc/ssh/sshd_config` and change **PasswordAuthentication no** to **PasswordAuthentication yes**
3) restart ssh service: `sudo systemcel restart ssh`
4) ssh into board: `ssh mendel@192.168.100.2`, i think this should also works: `ssh mendel@vexing-goose` (password is "mendel"). Also, you can use the wifi address instead of 192.168.100.2 if you don't want to connect the board to your computer all the time.

That should allows you to ssh into it, this is actually my prefered method. Although I'm still not sure why mdt isn't working in the first place for both of you guys o_0

Charlygnusa, 2020-08-27:
@Namburger Thanks for the help

but I tried to connect to the board via the serial console, it doesn't work, every time I run `screen / dev / ttyUSB0 115200`, it always `cannot exec / dev / ttyUSB0: No such file or directory`, even though the serial cable is connected, and in My ubuntu log shows `device not accepting address 20, error -71`, what should I do?

Namburger, 2020-08-27:
@Charlygnusa that sounds like your host machine did not detect the board, weird, I've never seen this. 
Could you try turning off the board, plug it back in (both serial and usb-c cables are necessary for this task: USB-A to USB-micro-B cable, USB-A to USB-C cable)
Then check 
```
dmesg | grep ttyUSB
```

This should have prompt you in the serial console just as the first time you flashed the board

Charlygnusa, 2020-08-28:
@Namburger sorry I can only reply now and thank you for always helping me

Finally I know the problem, it turns out the cable is not working, when I replace the cable that I have, the screen is running, but I have a problem with `screen is terminating`, I will find out first to fix it, then I will try the method you give

Thank you for the help

Namburger, 2020-08-28:
No problems! 
`screen is terminating` is usually an easy fix, unplug and replug :p

Charlygnusa, 2020-08-28:
@Namburger hmm I have tried it but it doesn't work, then I run `screen -ls`, and it shows No Sockets found in `/run/screen/S-mypc`, do you know why this happened?

Namburger, 2020-08-28:
@Charlygnusa not sure :/ 
maybe try using screen with sudo or try it on a different host computer?

Charlygnusa, 2020-08-28:
I tried with sudo but it only appears blank, maybe I will try it with a different host computer, thanks for the help

epikjjh, 2020-09-26:
Hello, I have the same problem here. (OS: Ubuntu 18.04)
when I strace with mdt shell, I found those results:

ioctl(1, TIOCGWINSZ, {ws_row=55, ws_col=101, ws_xpixel=0, ws_ypixel=0}) = 0
stat("/usr/lib/python3.6/encodings", {st_mode=S_IFDIR|0755, st_size=20480, ...}) = 0
stat("/usr/lib/python3.6/encodings/idna.py", {st_mode=S_IFREG|0644, st_size=9170, ...}) = 0
stat("/usr/lib/python3.6/encodings/idna.py", {st_mode=S_IFREG|0644, st_size=9170, ...}) = 0
openat(AT_FDCWD, "/usr/lib/python3.6/encodings/__pycache__/idna.cpython-36.pyc", O_RDONLY|O_CLOEXEC) = 9
fstat(9, {st_mode=S_IFREG|0644, st_size=5765, ...}) = 0
lseek(9, 0, SEEK_CUR)                   = 0
fstat(9, {st_mode=S_IFREG|0644, st_size=5765, ...}) = 0
read(9, "3\r\r\n\223\236\21_\322#\0\0\343\0\0\0\0\0\0\0\0\0\0\0\0\5\0\0\0@\0\0"..., 5766) = 5765
read(9, "", 1)                          = 0
close(9)                                = 0
stat("/home/jsense/.local/bin", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
stat("/usr/lib/python3.6", {st_mode=S_IFDIR|0755, st_size=20480, ...}) = 0
stat("/usr/lib/python3.6/stringprep.py", {st_mode=S_IFREG|0644, st_size=12917, ...}) = 0
stat("/usr/lib/python3.6/stringprep.py", {st_mode=S_IFREG|0644, st_size=12917, ...}) = 0
openat(AT_FDCWD, "/usr/lib/python3.6/__pycache__/stringprep.cpython-36.pyc", O_RDONLY|O_CLOEXEC) = 9
fstat(9, {st_mode=S_IFREG|0644, st_size=10030, ...}) = 0
lseek(9, 0, SEEK_CUR)                   = 0
fstat(9, {st_mode=S_IFREG|0644, st_size=10030, ...}) = 0
read(9, "3\r\r\n\223\236\21_u2\0\0\343\0\0\0\0\0\0\0\0\0\0\0\0\213\2\0\0@\0\0"..., 10031) = 10030
read(9, "", 1)                          = 0
close(9)                                = 0
brk(0x1948000)                          = 0x1948000
brk(0x196a000)                          = 0x196a000
socket(AF_INET, SOCK_STREAM|SOCK_CLOEXEC, IPPROTO_IP) = 9
connect(9, {sa_family=AF_INET, sin_port=htons(22), sin_addr=inet_addr("192.168.100.2")}, 16

--> stucked for a while then,

 ) = -1 ETIMEDOUT (Connection timed out)
write(1, "Couldn't establish ssh connectio"..., 92Couldn't establish ssh connection to device: socket error: [Errno 110] Connection timed out
) = 92
getsockname(9, {sa_family=AF_INET, sin_port=htons(41900), sin_addr=inet_addr("0.0.0.0")}, [16]) = 0
getpeername(9, 0x7ffc6743af10, [16])    = -1 ENOTCONN (Transport endpoint is not connected)
close(9)                                = 0
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f74a1712fd0}, {sa_handler=0x630060, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f74a1712fd0}, 8) = 0
munmap(0x7f749e903000, 262144)          = 0
sigaltstack(NULL, {ss_sp=0x15a4980, ss_flags=0, ss_size=8192}) = 0
sigaltstack({ss_sp=NULL, ss_flags=SS_DISABLE, ss_size=0}, NULL) = 0
futex(0x7f74a00ce948, FUTEX_WAKE_PRIVATE, 2147483647) = 0
fstat(0, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
fstat(0, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
fstat(0, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
exit_group(1)                           = ?
+++ exited with 1 +++

I reflashed coral image several times, but I got same results. I think the problem results from ubuntu maybe.