f0cal/google-coral #80: Getting started, books, basic scripts, etc. (classification on a video file)

mikelittlerock, 2020-07-22:
Greetings everyone. I have a Coral dev board as well as the USB model and eager to get started. I have found some samples where people classified live videos but I am trying to get a basic script as a starting point. basically just open file "video_sample1.mpg (or avi etc) and run classification and return basic results.. "Human 84%" or "Deer 98%" etc.. 

I am also interested in any sites, books, etc. that are good for people new to Coral and tensorflow.

Namburger, 2020-07-23:
@mikelittlerock 
How about this example? https://github.com/google-coral/examples-camera/tree/master/opencv
It's object detection, only 2 files and very easy to read :)

mikelittlerock, 2020-07-27:
@Namburger Thanks but this one is also pulling images via local Pi camera, correct? im trying to find an example where it reads a MPG or AVI stored locally.

Namburger, 2020-07-27:
@mikelittlerock You can easily change [this line](https://github.com/google-coral/examples-camera/blob/master/opencv/detect.py#L94) from:
```
cap = cv2.VideoCapture(args.camera_idx)
```
to 
```
cap = cv2.VideoCapture('name_of_video_file.avi')
```

to read frames from a video file instead! 
Here is the proper documentation from opencv: https://opencv-python-tutroals.readthedocs.io/en/latest/py_tutorials/py_gui/py_video_display/py_video_display.html#playing-video-from-file

Cheers

closed 2020-07-27 by Namburger