f0cal/google-coral #81: Command of Coral Dev board ??

IvanChavR, 2020-07-29:
* Doc you were trying to follow: <please insert>
* Your host OS: <please insert>
* Your Python3 version: <please insert>

# What I ran

# What the docs said should happen

# What actually happened

Namburger, 2020-07-29:
@IvanChavR What do you mean by commands?
We usually just use the `time` module to measure the inference time, similar to this: https://github.com/google-coral/tflite/blob/master/python/examples/classification/classify_image.py#L110-L112

IvanChavR, 2020-07-31:
Sorry for the question, I am a beginner with the coral, what I mean is, Are there commands to see the  performance of TPUs?
something similar such as nvidia-smi to check the GPUs

Namburger, 2020-07-31:
@IvanChavR nothing to be sorry about :)
I just wasn't sure what you mean. So I looked up `nividia-smi` and it seems like a lot of info that we currently don't expose. The nvidia's gpu is more general and can be used for different things, while our GPU is specifically for processing tensors. Unfortunately, there is not much specs to expose at this time. 
However, the most relevant info to benchmark the performance of the tpu is still via measuring duration of inference, which I mentions above!

IvanChavR, 2020-08-04:
Thank you @Namburger for the answer

Bhanuchander210, 2020-12-01:
Looking for the same :)
[How to fetch TPU utilization details in coral developer board using tflite_runtime or linux command](https://stackoverflow.com/questions/65086544/how-to-fetch-tpu-utilization-details-in-coral-developer-board-using-tflite-runti)

iaverypadberg, 2022-05-13:
I'm also looking for a way to get more in depth info on the TPUs utilization metrics(temp & %utilization).