f0cal/google-coral #83: Issue running image detection (new user)

Dave592309, 2020-08-07:
* Doc you were trying to follow: 
* Your host OS: Mac OS Mojave
* Your Python3 version: 3

# What I ran
Here is the video: https://www.loom.com/share/6a9209cb38b34cf88c58a57f426653d2

# What the docs said should happen
https://coral.ai/docs/dev-board/camera/#using-an-hdmi-monitor

# What actually happened
Can't get the model to run.

Namburger, 2020-08-07:
@Dave592309 Is this a classification model or detection model? 
From what I understand, you basically moved the copied the model and label file to the dev board, just having problem running it?

Namburger, 2020-08-07:
@Dave592309 just took a look at the project, it looks like a classification model, I'm assuming that you have a coral dev board and also a coral's camera attached, then you should be able to run it like this:

```
edgetpu_classify --model path_to_your_model --label path_to_your_label
```
Basically following this demo: https://coral.ai/docs/dev-board/camera/#using-an-hdmi-monitor

Cheers

closed 2020-08-18 by Namburger